ARMmbed/mbed-os #4080: USBSerial timeout before connection ready

Split7fire, 2017-03-30:
### Description
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
STM32F407 Discovery board

**Toolchain:**
GCC_ARM (PlatformIO)

**mbed-cli version:** 
5.24

**Expected behavior**
After upload USBserial example to board nearly instant access to usbserial serial console (An half-year ago this behavior was true).

**Actual behavior**
After successful sketch upload, ~ 3 minutes I can not open port `could not open port u'/dev/ttyACM0': [Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'` After ~ minutes serial connection began to open normally and I can read/write to serial console.

**Steps to reproduce**
Compile USBSerial example and upload to board, check usbserial console

0xc0170, 2017-03-30:
cc @bcostm @adustm @LMESTM @jeromecoutant

Split7fire, 2017-03-31:
What more debug info can I provide?

bcostm, 2017-03-31:
cc @jamike

cmonr, 2018-01-23:
Closing due to inactivity. If this is still an issue, feel free to reopen it up, or create a new issue.

closed 2018-01-23 by cmonr