ARMmbed/mbed-os #4082: EFM32: Sleep function does not cause the MCU to go to a sleep mode

seppestas, 2017-03-30:
In mbed-os 5 and later versions of the mbed 2 SDK (aka mbed-dev), the `sleep()` function does not cause EFM32 MCUs to sleep properly, even when no sleep levels have been blocked.

When testing on the EFM32 Happy Gecko 322 using the EFM32HG_STK3400 target the MCU still consumes 3.49mA at 3v3 when calling `sleep` in a loop. I was able to reproduce this in:
- mbed-os 5 at f4864dc6429e1ff5474111d4e0f6bee36a759b1c
- the pre-compiled mbed lib at 093f2bd7b9eb
- mbed-dev revision d5399cc887bb

I tested both on the STK3400 devkit and a custom breakout board for the EFM32HG322 MCU.

Using mbed-dev at revision d6373279a6f1 using a custom build script and the mbed online compiler I did not have this issue yet. I am still trying to build this revision using the mbed tools.

I still plan to test on other EFM32 MCU's, but since a lot of code is shared I'm guessing the results will be similar.

---------------------------------------------------------------
## Bug

**Target**
EFM32

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
1.0.0

chrissnow, 2017-03-30:
I had a similar problem with the PG, it's to do with sleep being disabled in debug builds.
```c++
#ifdef NDEBUG
#if DEVICE_SLEEP
    hal_sleep();
#endif /* DEVICE_SLEEP */
#endif /* NDEBUG */
```
if you define NDEBUG it should sleep correctly.

chrissnow, 2017-03-30:
Also you may need to power cycle the processor to get it out of debug mode after programming or it still wont sleep.

seppestas, 2017-03-31:
Thanks @chrissnow, this seemed to be the problem indeed! I did not know about the `hal_sleep` functionality. IIRC this was not there in mbed 2 yet.

It would be nice to have this sort of stuff documented on e.g the [Power-Management cookbook](https://developer.mbed.org/cookbook/Power-Management). Notice the example won't work in the Debug build (which seems to be the default).

bulislaw, 2017-03-31:
Actually I think we need to update that, and made the board **not** go to sleep only in the `debug` profile, as opposed to the default `develop`and the `release`. I'll try to cook something up later today. And update the build profile docs.

seppestas, 2017-03-31:
@bulislaw great, thanks.

Let me know if I can be of help, I'm currently looking into the power consumption of leuart on the EFM32 Happy Gecko and I'm building up some interesting scope images.

bulislaw, 2017-03-31:
https://github.com/ARMmbed/mbed-os/pull/4097 changes to sleep

cmonr, 2018-06-15:
Closing as issue appear to be fixed.

closed 2018-06-15 by cmonr