ARMmbed/mbed-os #4083: mbed test - specify board

andreaslarssonublox, 2017-03-30:
### Description
- Type:  Enhancement
- Related issue: None
- Priority: Minor

----------------------------------------------------------------
## Enhancement
It would be very useful to be able to specify what physical board to use when running 'mbed test'.
By doing so the user can run multiple tests in parallel for example for all three toolchains.
The board can be identified by it's target_id, perhaps in some short form or COM port.
The advantage with the target_id is that one might have several boards with different setups and need to run against a specific board.

An alternative might be to be able to run all toolchains in one go.

0xc0170, 2017-03-31:
cc @theotherjimmy @screamerbg

screamerbg, 2017-03-31:
Cc @bridadan