ARMmbed/mbed-os #4084: Exported project build fails in SW4STM32

vargham, 2017-03-30:
### Description
I tried System Workbench for STM32. I exported MBED blinky from online IDE and build failed: missing declaration of MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE
The define is in mbed_config.h, this file is present in the project root.
- Type: Bug
- Priority: Major
```
make all 
Building file: ../main.cpp
Invoking: MCU G++ Compiler
%cd%
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DMBED_BUILD_TIMESTAMP=1490892052.03 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_NUCLEO_F103RB -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_STM32F103RB -I"D:/_dev/_workspaces/sws/mbed_blinky/." -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/." -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TARGET_STM" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TARGET_STM/TARGET_STM32F1" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB/device" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TARGET_STM/TARGET_STM32F1/device" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/TARGET_NUCLEO_F103RB/TOOLCHAIN_GCC_ARM" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/drivers" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/hal" -I"D:/_dev/_workspaces/sws/mbed_blinky/mbed/platform" -I"D:/_dev/_workspaces/sws/mbed_blinky/" -O0 -g3 -Wall -fmessage-length=0 -ffunction-sections -c -fno-exceptions -fno-rtti -MMD -MP -MF"main.d" -MT"main.o" -o "main.o" "../main.cpp"
In file included from d:\_dev\_workspaces\sws\mbed_blinky\mbed\mbed.h:80:0,
                 from ../main.cpp:1:
d:\_dev\_workspaces\sws\mbed_blinky\mbed\drivers/Serial.h:70:70: error: 'MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE' was not declared in this scope
     Serial(PinName tx, PinName rx, const char *name=NULL, int baud = MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE);
                                                                      ^
In file included from d:\_dev\_workspaces\sws\mbed_blinky\mbed\mbed.h:87:0,
                 from ../main.cpp:1:
d:\_dev\_workspaces\sws\mbed_blinky\mbed\drivers/RawSerial.h:64:50: error: 'MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE' was not declared in this scope
     RawSerial(PinName tx, PinName rx, int baud = MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE);
                                                  ^
make: *** [main.o] Error 1
```
---------------------------------------------------------------
## Bug

**Toolchain:**
Online exporter

**Toolchain version:**
Current online exporter (2017-03-30)

**Expected behavior**
Exported project should be built as-is.

**Actual behavior**
Build fails, missing define MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE

**Steps to reproduce**
Create project "mbed_blinky" for an STM32 target and export it to offline SW4STM32. Build will fail.

vargham, 2017-03-30:
Workaround: Include mbed_config.h in RawSerial.h and in Serial.h

theotherjimmy, 2017-03-30:
The exporter should be doing a "-include mbed_config.h", but I don't see it.

ronanpaixao, 2017-06-03:
This is pretty much the same as #3606

JojoS62, 2017-06-26:
This include is still missing. Adding "${ProjDirPath}/mbed_config.h" manualy to Project Properties/ C/C++ Build / Settings / C an C++ compiler includes' will fix it temporarily.

adbam, 2017-06-29:
Can confirm this issue on a Nucleo_F303RE. When I include mbed_config.h manually the compile succeeds. 

But the compilation is still not correct, as I'm trying with the ticker example. When I compile from the online compiler my LED flashes as expected, but when I compile from SW4STM32 - and the compilation succeeds - nothing happens on my target. 
Is there a bigger problem in the export feature? Anyone knows a workaround for this problem for now?

PR #4779 (2017-09-22): Fixed sw4stm32 exporter
Notes:
Fixes issues with sw4stm32 exporter

## Description
Added mbed_config.h as -included, fixes #4084. 

Added exclude directory generation, fixes #4091 (see theotherjimmy post). 

Added -O2 to GCC compiler because e.g. mbed-os/rtos/rtx5/TARGET_CORTEX_M/rtx_delay.c needs it. 

Exporter generated broken project file if there was quotes in toolchain symbols, fixed that also.

## Status
**READY**

## Todos
- [ ] Tests
- [ ] Documentation

## Deploy notes

## Steps to test or reproduce
closed 2017-09-22 by 0xc0170