ARMmbed/mbed-os #4086: ChaNfs - sector size

chrissnow, 2017-03-30:
### Description
- Type: Bug & Enhancement 
- Priority:  Minor

---------------------------------------------------------------
## Bug
I have been implementing a block device driver for the Adesto AT45 (ex atmel) DataFlash devices and came across an issue with memory corruption.
The DataFlash by default has sector sizes in multiple of 264 bytes depending on density but can configured in "binary mode" which has sectors in multiples of 256, you just lose some space.
I have been using a 16Mbit chip so has 512\528 bytes per sector.

get_erase_size() etc was implemented to report 528 bytes.
all seemed to be working fine but hard faults were appearing in odd places and it turns out that the buffer that was passed to the BD was actually only 512B but the size requested was 528B so I was writing over over other things, no errors were thrown.

I found the following.
in ff.cpp
`#if _MAX_SS != _MIN_SS		/* Get disk sector size */
	if (disk_ioctl(pdrv, GET_SECTOR_SIZE, &SS(fs)) != RES_OK || SS(fs) > _MAX_SS || SS(fs) < _MIN_SS)
		return FR_DISK_ERR;
#endif`

in ffconf.h
`#define	_MIN_SS		512
#define	_MAX_SS		512`

so the sector sizes are in fact not checked if a fixed size of 512b is set here (_MIN_SS	=_MAX_SS)

Increasing _MAX_SS fixed the issues.

now comes some questions..

- Is there any advantage of supporting more than 512 bytes per sector?
if so _MAX_SS needs increasing to 1024 or bigger, obviously this increases the memory footprint for the buffer.
- How does the implementation deal with the 4096 that the SPIFBlockDevice reports? is it just using the first 512B per sector?
-Should the FS implementation check that the reported sector size is <=  _MAX_SS since ChaNfs doesn't? (if _MIN_SS =_MAX_SS)
- Could _MAX_SS be exposed or configured in a use friendly way?

I need to tidy up the code a bit and see what the opinion is here on 512\528 then I will submit a PR.

DataFlash could work nicely for people since it has small erase sizes so works with a small memory footprint without wasting 3/4 of your storage.

geky, 2017-03-30:
Hi @chrissnow! Have you tried this pr: https://github.com/ARMmbed/mbed-os/pull/3972

You are right, the current master of mbed OS breaks terribly and without warning when erase size != 512. I'm hoping https://github.com/ARMmbed/mbed-os/pull/3972 fixes this.

Out of curiousity, did the fatfs work find with 528B blocks?

As for which block size to report, I would stick with 528B. FTL layers may use the odd space to store metadata about blocks and present a more conventional power of 2 blocks size. There may be some rough support at the beginning, but applications can always round down.

DataFlash support would be cool and sounds like it would fit well.

chrissnow, 2017-03-30:
@geky I hadn't noticed that PR, Looks to be exactly what I need, I will try it tomorrow.
I haven't extensively tested it yet but I didn't need make any changes in fatfs to get it to format and run the basic file tests.

I like the use of heap, my original concern with increasing the default to 4k was that it would make it unusable for ram constrained devices.

I would think this would benefit other non mbed people so it would be nice if they merged your changes.

another thing that could do with some thought is..

```c++
/*---------------------------------------------------------------------------/
/ Drive/Volume Configurations
/---------------------------------------------------------------------------*/

#define _VOLUMES	1
/* Number of volumes (logical drives) to be used. */`
```
I have SD + DataFlash on a board and this needs to be increased to 2, admittedly it's not likely people will have 2 mounts but I wonder how much work it would be to make that dynamic too?

geky, 2017-03-31:
Looking into it, I can't think of a reason to not set `_VOLUMES` to the max of 9 (limited [here](https://github.com/ARMmbed/mbed-os/blob/9844a390d9e70381d654d139b6eec821a3953c5c/features/filesystem/fat/ChaN/ff.cpp#L445)). The only cost is 9 pointers worth of space ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L131)).

It _should_ all just work. Feel free to create a pr, or I can get to it when I can test >1 fatfs.

chrissnow, 2017-04-25:
@geky here is the DataFlash block device driver I have been working on.
https://github.com/chrissnow/DataFlashDevice

it's based on your SFDP driver and some help from here
https://developer.mbed.org/users/stjo2809/code/AT45/

I haven't finished testing #3972 yet or `_VOLUMES` though that is a quick test as I have this and SD on a board.