ARMmbed/mbed-os #4091: Exported mbed-os project build fails in SW4STM32

vargham, 2017-03-31:
### Description
- Type: Bug
- Priority: Major
After applied workarounds in `#4075` and in `#4084` the mbed blinky example compiles fine offline. When I changed mbed lib to mbed-os lib, the blinky compiles in the online IDE but fails in offline SW4STM32 IDE. Error: storage_volume_manager.h: No such file or directory The file presents in another directory, so I can move it to the right place, but after all other .h files also missing below FEATURE_STORAGE directory.

```
Building file: ../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp
Invoking: MCU G++ Compiler
%cd%
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_PORTOUT=1 -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DMBED_BUILD_TIMESTAMP=1490898149.04 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTARGET_NUCLEO_F103RB -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_STM32F103RB -O0 -g3 -Wall -fmessage-length=0 -ffunction-sections -c -fno-exceptions -fno-rtti -MMD -MP -MF"mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.d" -MT"mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o" -o "mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o" "../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp" @"mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/includes.args"
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp:18:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
compilation terminated.
make: *** [mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o] Error 1
```

The missing storage_volume_manager.h file is in another directory:
`../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/storage-volume-manager/`

---------------------------------------------------------------
## Bug

**Target**
STM32

**Toolchain:**
System Workbench for STM32 with GCC_ARM

**Expected behavior**
Exported project should be built as-is.

**Actual behavior**
Build fails with missing file error.

**Steps to reproduce**
Create mbed blinky example for STM32 target, change mbed lib to mbed-os lib, export to System Workbench for STM32 and try to build. (You have to apply workarounds mentioned before to reach this point.)

0xc0170, 2017-03-31:
cc @bcostm @adustm @LMESTM @jeromecoutant

adustm, 2017-04-03:
Hello
It looks like mbed-os lib her tries to compile the Storage feature from mbed-os, for NUCLEO_F103RB, but it should not, as this feature is not part of the targets.json for NUCLEO_F103RB.

@theotherjimmy , would you have a clue for that ?
Kind regards

leeehui, 2017-05-11:
Hi all,
I solved this problem by adding the following include path to properties->C/C++ Build->Settings->MCU GCC Compiler->Includes:
"${ProjDirPath}/mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/storage-volume-manager"

theotherjimmy, 2017-05-11:
Actually, the problem is a bit different from that. The exporter ignores which files it should be compiling, and compiles all of them. However, it does heed the include directories. This disconnect is what's causing the problem here. @adustm, if you have time, I would appreciate your help on getting the SW4STM32 exporter to understand mbed-os  projects better.

StefanHri, 2017-05-26:
I have the same problem. Do you have a solution?

adustm, 2017-05-29:
Hello, we are discussing this issue internally. I hope we will be able to help soon.

noutram, 2017-06-26:
Any updates on this issue?

PR #4779 (2017-09-22): Fixed sw4stm32 exporter
Notes:
Fixes issues with sw4stm32 exporter

## Description
Added mbed_config.h as -included, fixes #4084. 

Added exclude directory generation, fixes #4091 (see theotherjimmy post). 

Added -O2 to GCC compiler because e.g. mbed-os/rtos/rtx5/TARGET_CORTEX_M/rtx_delay.c needs it. 

Exporter generated broken project file if there was quotes in toolchain symbols, fixed that also.

## Status
**READY**

## Todos
- [ ] Tests
- [ ] Documentation

## Deploy notes

## Steps to test or reproduce
closed 2017-09-22 by 0xc0170