ARMmbed/mbed-os #4096: STM32F411 mbed blinky build failed in EmBitz

vargham, 2017-03-31:
### Description
Exported mbed blinky for STM32F411 to EmBitz. Build failed. Same project export works fine with STM32F103. Plain mbed lib, no OS.
Note: Have to manually copy mbed_config.h as described here: https://github.com/ARMmbed/mbed-os/issues/4075
- Type: Bug
- Priority: Major

```
-------------- Build: mbed_blinky in mbed_blinky ---------------
Compiling: main.cpp
Linking executable: .\build\mbed_blinky.elf
mbed\TARGET_NUCLEO_F411RE\TOOLCHAIN_GCC_ARM\libmbed.a(mbed_wait_api_no_rtos.o): In function `wait':
mbed_wait_api_no_rtos.c:(.text.wait+0x0): multiple definition of `wait'
c:/program files (x86)/embitz/1.11/share/em_armgcc/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/softfp\libc.a(lib_a-syswait.o):syswait.c:(.text.wait+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
Process terminated with status 1 (0 minutes, 3 seconds)
0 errors, 0 warnings (0 minutes, 3 seconds)
```

---------------------------------------------------------------
## Bug

**Target**
STM32F411

**Toolchain:**
Online IDE exporter, EmBitz, GCC_ARM

**Expected behavior**
Exported project should be built as-is.

**Actual behavior**
BUild failes with error.

**Steps to reproduce**
Create mbed blinky example for STM32F411 target, export to EmBitz and try to build.

0xc0170, 2017-04-03:
cc @EmBitz @theotherjimmy

theotherjimmy, 2017-04-03:
This might be a targets.json mis-configuration. Does in `mbed compile` correctly?

theotherjimmy, 2017-04-03:
Also, I can't find the target `STM32F411` could you include your target in the description or post it in a comment below?

theotherjimmy, 2017-04-03:
@EmBitz I don't think you need to worry about this one. I'll let you know if I need your help after I root-cause it.

vargham, 2017-04-03:
Yes, I used this board https://developer.mbed.org/platforms/ST-Nucleo-F411RE/
The online compilation is fine.

theotherjimmy, 2017-04-03:
Thanks for the clarification @vargham I'm going to do this export myself and see what's what.

closed 2018-03-28 by ChiefBureaucraticOfficer