ARMmbed/mbed-os #4099: Add I2C_SDA[0,1,2] and I2C_SCL[0,1,2] definitions to LPC1768 PinNames.h

wdwalker, 2017-03-31:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
Summary:

The main reason is that the I2C_[SCL,SDA] definitions for the LPC1768 don't map to the I2C used by the LPC1768 application board. In fact, they don't even seem to map to any pins on the development board.

Detail: 

Hopefully I'm not blowing my analysis here. Looking at the NXP LPC1768 user manual, the diagram at https://developer.mbed.org/platforms/mbed-LPC1768, and LPC1768's [PinNames.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768/PinNames.h), we get:

- I2C0: 
-- SDA0 = P0_27 = no PinName
-- SCL0 = P0_28 = no PinName

- I2C1: 
-- SDA1 =P0_0 = p9
-- SCL1 = P_1 = p10

- I2C1 (alternate):
-- SDA1 = P0_19 = no PinName
-- SCL1 = P_20 = no PinName

- I2C2:
-- SDA2 = P0_10 = p28
-- SCL2 = P0_11 = p27

LPC1768's PinNames.h is defining I2C_SDA and I2C_SCL to be for I2C0, which does not seem useful for us: I don't see them being mapped to any pins on the development board. As a result, you end up seeing many people writing I2C code for the LPC1768 referring to p27 and p28.

**Suggested enhancement**

1. If a convention has not been established for I2C naming, it would be great to have one. I'd propose using I2C_SDA and I2C_SCL for the "default" and then I2C_SDA[0,1,2,...] and I2C_SCL[0,1,2,...] for specific ones.

2. Add these specific defs to LPC1768's PinNames.h (moving NC above as appropriate):

    I2C_SCL0 = NC,
    I2C_SCL1 = NC,
    I2C_SCL1 = p10,
    I2C_SDA1 = p9,
    I2C_SCL2 = p28,
    I2C_SDA2 = p27,

3. I'd also suggest changing the LPC1768's PinNames.h I2C_SCL and I2C_SDA to values that are bound to pins on the LPC1768 application board:

    I2C_SCL = I2C_SCL2,
    I2C_SDA = I2C_SDA2,

**Pros**

- A consistent naming convention across targets would be nice.
- We will end up with usable macros instead of ones that are ignored.

**Cons**

- The part about changing the existing definitions of LPC1768's PinNames.h I2C_SCL and I2C_SDA may be risky because they've been that way in the file since at least 2014. If my analysis is correct, however, I can't see how anyone would be actually using these.

Again - apologies if my analysis is wrong.

theotherjimmy, 2017-03-31:
Hello again @wdwalker. It looks like you have done all of the research again! Bravo! I recommend submitting these sorts of things in PR form and tagging @0xc0170. He will tag the additional parties that originally wrote or are currently maintaining the software you change. Doing things this way will reduce the time it takes to get feedback on and adopt the changes you propose.

theotherjimmy, 2017-03-31:
Wonderful writeup BTW. If this was a PR, the only thing that would need to change is that you would have commits attached.

wdwalker, 2017-03-31:
Thanks @theotherjimmy. Quick question - does mbed culture like an issue for every PR, or can PRs just stand by themselves? (different open source cultures behave differently - what is acceptable to some is completely blasphemous to others)

theotherjimmy, 2017-03-31:
>Quick question - does mbed culture like an issue for every PR, or can PRs just stand by themselves? 

We have a culture of just submitting PRs with descriptions of the problem they are solving with or without an issue. PRs can stand by themselves.

>(different open source cultures behave differently - what is acceptable to some is completely blasphemous to others)

:laughing:

closed 2017-04-19 by sg-