ARMmbed/mbed-os #4105: EFM32: Synchronous serial write does not clear interrupts

seppestas, 2017-04-03:
## Bug

**Target**
EFM32 (tested on EFM32HG_STK3400 and EFM32LG_STK3600)

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
1.0.0

**meed-os sha:**
f4864dc64

**Expected behavior**
When combining synchronous writes to serial (i.e `Serial.printf`, `Serial.puts` or `Serial.putc`) with asynchronous writes (i.e `Serial.write`) the asynchronous write is expected to work properly (assuming no race-conditions). This means the system is supposed to be able to go to a low power sleep mode when the data has been sent.

**Actual behavior**
When the data has been sent, the MCU does not go to a proper low-power sleep mode or something else is causing a higher-than-expected power consumption. Following scope image is taken from an EFM32HG_STK3400 dev board sending data using `Serial.printf` at boot and echoing received lines using Serial.write using the Low Energy UART:
[image]

The same code on the EFM32LG_STK3600 target causes the same issue (with a slightly higher power consumption):
[image]

When not sending the initial data using `Serial.printf` the issue does not occur (on EFM32LG_STK3600 board):
[image]

And when using only synchronous writes the issue does not occur either:
[image]

**Steps to reproduce**
- Program following code on an EFM32 target, [make sure to compile using release target](https://github.com/ARMmbed/mbed-os/issues/4082):
``` cpp
#include "mbed.h"

#define RX_BUFFER_SIZE 10

DigitalOut led(LED0);
Serial leuart(PD4, PD5);
event_callback_t serialEventCb;
uint8_t rx_buf[RX_BUFFER_SIZE];

void serialCb(int events) {
    unsigned char i;
    if(events & SERIAL_EVENT_RX_CHARACTER_MATCH) {
        for(i = 0; i < RX_BUFFER_SIZE; i++) {
            if(rx_buf[i] == '\n') break;
        }
    } else if (events & SERIAL_EVENT_RX_COMPLETE) {
        i = RX_BUFFER_SIZE - 1;
    } else {
        led = 1;
    }
    leuart.write(rx_buf, i+1, 0, 0);
    leuart.read(rx_buf, RX_BUFFER_SIZE, serialEventCb, SERIAL_EVENT_RX_ALL, '\n');
}

int main()
{
    leuart.printf("Hello, world\r\n");

    serialEventCb.attach(serialCb);
    leuart.read(rx_buf, RX_BUFFER_SIZE, serialEventCb, SERIAL_EVENT_RX_ALL, '\n');

    while(true) {
        sleep();
    }
    return 0;
}
```
- Measure the current draw from the battery, make sure to put the current source select switch to `BAT` and to disconnect the debugger.
- In scope images, current drawn was measured using a [µCurrent](https://www.eevblog.com/projects/ucurrent/) connected to Channel 1. Channel 2 was connected to pin `PD4`. The battery power was supplied by a 3V3 power supply.

## Notes
- 1mA in the scope images is actually 1µA. This is caused by the output unit of the µCurrent being set to 1mV/µA and the Scope using a 1X probe configuration (sadly X0.001 is not available on the Rigol DS1054).
- The range of the µCurrent is limited to 1.5mV, so the maximum current that can be measured is 1500µA or 1.5mA.

seppestas, 2017-04-03:
I tried disabling the interrupts when the sync write completes. This seems to fix the issue, but I only implemented / tested it on the leuart. I might do some more testing and provide a PR when I have time (end of this week).

0xc0170, 2017-04-03:
cc @stevew817 @akselsm

stevew817, 2017-04-04:
Thank you @seppestas for a very detailed and exemplary bug report! I'll have a look at it when I can.

cmonr, 2018-06-14:
Closing since no activity has occured in months.

Feel free to open a new issue if this is still a problem.

closed 2018-06-14 by cmonr

seppestas, 2018-06-14:
Sadly this issue didn’t fix itself. I believe closing it without fixing it reduces visibility, preventing other people who encounter similar problems from finding out what is happening.

As far as I can tell this issue still exists, and I already found a lot of similar issues on other platforms.

0xc0170, 2018-06-15:
> Sadly this issue didn’t fix itself. I believe closing it without fixing it reduces visibility, preventing other people who encounter similar problems from finding out what is happening.

Can be reopened, we would appreciate an update as this is more than a year old. Thanks for the feedback.

@ARMmbed/team-silabs can you review?

seppestas, 2018-06-15:
I believe the problem is caused by sync serial writes not disabling/clearing interrupts. Sadly fixing it was beyond the scope of my project and I didn't have enough time to spare.

On top of that I only have a couple of EFM32 targets available for testing, so I don't think I will be able to fix this on my own, even though I think the fix is simple.

This issue took me a lot of time to figure out and has a very clear cause on a well defined set of targets, I believe it has a lot of value for other people having similar issues with low power operaton, if only just to tell them not to use sync and async writes in the same program.