ARMmbed/mbed-os #4108: Timer/Ticker/RTC/Thread issue on NRF51_DK and derived targets. Porting to custom board docs?

smarek, 2017-04-04:
### Description
- Type: Question
- Related issue: #4026 #3857 #4016 #2598
- Priority: Major

---------------------------------------------------------------
## Question

**Target**
NRF51_DK / NRF51_CUSTOM

**Toolchain:**
GCC_ARM

**Toolchain version:**
(GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

**mbed-cli version:** 
(1.0.0)

**meed-os sha:**
50b3418e4 Merge pull request #4025 from ARMmbed/release-candidate

**Expected behavior**
Should stabily run, blinking every 1 second on led1

**Actual behavior**
Will halt in range of 20 seconds and 5 minutes, after a while, will restart itself

**Steps to reproduce**

Code
```cpp
#include "mbed.h"

Thread led_blink1;
DigitalOut led1(P0_8); // P0_21 for NRF51_DK

void topledTick() {
	while(1){
		led1 = !led1;
		wait(1);
	}
}

int main() {
	led_blink.start(callback(topledTick));
	while (1) {	}
}
```
---------------------------------------------------------------

We've got NRF51_DK and custom board (with NRF51422 32K)

Using described code sample and build environment, this will result in code to hang (with leds on or off), and will restart itself after some time (which is much longer on custom boards than on DK)

I've also tried doing the same example using PwmOut on such pin, and not using Thread but Tickers, but there I've struggled with issues described in referenced issues. Also when using just PwmOut, I've tried to remove RTOS (using `mbed-os/rtos/*` line in .mbedignore), and that did not resolve the issue.

----------------------------------------------------------------

What I'd like to know

  1) Custom board uses DigitalOut on pin, which is part of Serial port (NRF51_DK PinNames.h RTS_PIN_NUMBER, CTS_PIN_NUMBER, TX_PIN_NUMBER, RX_PIN_NUMBER which should be LED1 through LED4). Is it possible that interrupt conflict (or something related to default mapping of those pins) will result in described problem?

  2) There is, afaik, no documentation, on how to create a custom target board, with different pin-out, which would inherit from existing target, such as NRF51_DK. What I've done is, I've copied, under new names, folders `./targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/TARGET_NRF51_DK` and `./targets/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_DK` to `TARGET_NRF51_CUSTOM` folders, in respective locations, and changed PinNames.h files (also I've copied and renamed target NRF51_DK within `targets.json` as `NRF51_CUSTOM`). Is this correct proceedure ?

  3) Simply changing the pin-out of target board is not possible, afaik, just by modifying `mbed_app.json` yotta config file, even though it's mentioned in DOCs that `config->hardware->pins` should be JSON-configurable ([source](https://docs.mbed.com/docs/mbed-hal-api/en/latest/api/md_pins.html)). Is it possible? If so, is there any more complete doc?

Seems to me, that NRF51422/NRF51822 is in terrible state when using mbed-os, or is it just me?

0xc0170, 2017-04-04:
cc @anangl @nvlsianpu

nvlsianpu, 2017-04-04:
Hi.
1) How to create custom board
> I've copied, under new names, folders ./targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/TARGET_NRF51_DK (...) to TARGET_NRF51_CUSTOM

Very good. I recommend to use this sources to build your target. Then in target.json your pattern for NRF51_CUSTOM is NRF51_DK.

I recommend to not using ./targets/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_DK is legacy target bases.

3) Customize your PinNames.h is the best way to define your own pinout. (Doc your mentioned is outdated)

smarek, 2017-06-21:
@pan- 

My custom target consists of:

- Copy of NRF51_DK, renamed to `CUSTOM_TARGET` and without SERIAL* and SPI* (see below)
- 'device.h' and 'PinNames.h' within `targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/CUSTOM_TARGET/`

I'm not exactly sure, where I could screw up so badly.
LED1-4 are phy pins 1,18,19,20
There is no UART/SERIAL/SPI interface exposed on my hardware, but I've commented out only UART, as it was in direct conflict with custom pin-out.

Testing with mbed-os-5.5.0-rc3 (3179b1cffddea0f0feed1dd551b0e7dc946c12f5)

*from targets.json*
```diff
diff --git a/targets/targets.json b/targets/targets.json
index 9e6eab404..9f3f971e6 100644
--- a/targets/targets.json
+++ b/targets/targets.json
@@ -2701,6 +2701,13 @@
         "release_versions": ["2", "5"],
         "device_name": "nRF51822_xxAA"
     },
+    "CUSTOM_TARGET": {
+        "supported_form_factors": ["ARDUINO"],
+        "inherits": ["MCU_NRF51_32K_UNIFIED"],
+        "device_has": ["ANALOGIN", "ERROR_PATTERN", "INTERRUPTIN", "LOWPOWERTIMER", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SLEEP"],
+        "release_versions": ["2", "5"],
+        "device_name": "nRF51822_xxAA"
+    },
     "NRF51_DONGLE": {
         "inherits": ["MCU_NRF51_32K_UNIFIED"],
         "progen": {"target": "nrf51-dongle"},
```

*PinNames.h*
```cpp
#ifndef MBED_PINNAMES_H
#define MBED_PINNAMES_H

#include "cmsis.h"

#ifdef __cplusplus
extern "C" {
#endif

typedef enum {
    PIN_INPUT,
    PIN_OUTPUT
} PinDirection;

#define PORT_SHIFT  3

typedef enum {
    p0  = 0,
    p1  = 1,
    p2  = 2,
    p3  = 3,
    p4  = 4,
    p5  = 5,
    p6  = 6,
    p7  = 7,
    p8  = 8,
    p9  = 9,
    p10 = 10,
    p11 = 11,
    p12 = 12,
    p13 = 13,
    p14 = 14,
    p15 = 15,
    p16 = 16,
    p17 = 17,
    p18 = 18,
    p19 = 19,
    p20 = 20,
    p21 = 21,
    p22 = 22,
    p23 = 23,
    p24 = 24,
    p25 = 25,
    p26 = 26,
    p27 = 27,
    p28 = 28,
    p29 = 29,
    p30 = 30,

    P0_0  = p0,
    P0_1  = p1,
    P0_2  = p2,
    P0_3  = p3,
    P0_4  = p4,
    P0_5  = p5,
    P0_6  = p6,
    P0_7  = p7,

    P0_8  = p8,
    P0_9  = p9,
    P0_10 = p10,
    P0_11 = p11,
    P0_12 = p12,
    P0_13 = p13,
    P0_14 = p14,
    P0_15 = p15,

    P0_16 = p16,
    P0_17 = p17,
    P0_18 = p18,
    P0_19 = p19,
    P0_20 = p20,
    P0_21 = p21,
    P0_22 = p22,
    P0_23 = p23,

    P0_24 = p24,
    P0_25 = p25,
    P0_26 = p26,
    P0_27 = p27,
    P0_28 = p28,
    P0_29 = p29,
    P0_30 = p30,

    LED1    = p8,
    LED2    = p9,
    LED3    = p10,
    LED4    = p11,

    BUTTON1 = p1,
    BUTTON2 = p18,
    BUTTON3 = p19,
    BUTTON4 = p20,

    //RX_PIN_NUMBER  = p11,
    //TX_PIN_NUMBER  = p9,
    //CTS_PIN_NUMBER = p10,
    //RTS_PIN_NUMBER = p8,

    // mBed interface Pins
    //USBTX = TX_PIN_NUMBER,
    //USBRX = RX_PIN_NUMBER,

    SPI_PSELMOSI0 = p25,
    SPI_PSELMISO0 = p28,
    SPI_PSELSS0   = p24,
    SPI_PSELSCK0  = p29,

    SPI_PSELMOSI1 = p13,
    SPI_PSELMISO1 = p14,
    SPI_PSELSS1   = p12,
    SPI_PSELSCK1  = p15,

    SPIS_PSELMOSI = p13,
    SPIS_PSELMISO = p14,
    SPIS_PSELSS   = p12,
    SPIS_PSELSCK  = p15,

    I2C_SDA0 = p30,
    I2C_SCL0 = p7,

    D0  = p12,
    D1  = p13,
    D2  = p14,
    D3  = p15,
    D4  = p16,
    D5  = p17,
    D6  = p18,
    D7  = p19,

    D8  = p20,
    D9  = p23,
    D10 = p24,
    D11 = p25,
    D12 = p28,
    D13 = p29,

    D14 = p30,
    D15 = p7,

    A0  = p17,
    A1  = p2,
    A2  = p3,
    A3  = p4,
    A4  = p5,
    A5  = p6,

    // Not connected
    NC = (int)0xFFFFFFFF
} PinName;

typedef enum {
    PullNone = 0,
    PullDown = 1,
    PullUp = 3,
    PullDefault = PullUp
} PinMode;

#ifdef __cplusplus
}
#endif

#endif
```

*device.h*

```cpp
#ifndef MBED_DEVICE_H
#define MBED_DEVICE_H

#include "objects.h"

#endif
```

pan-, 2017-06-21:
@smarek How out of curiosity what is your compilation command and did you clean your build folder after upgrading to mbed OS 5.5 ?

smarek, 2017-06-22:
I've cleaned by 'rm -rf BUILD' whole folder, also by checking out clean mbed-os repository and applying patch to provide CUSTOM_TARGET.
I compile with "mbed compile --verbose --profile mbed-os/tools/profiles/release.json"

And mbed global config looks like this, and no local/project config is defined
```
[mbed] Global config:
GCC_ARM_PATH=/home/user/opt/Nordic/toolchain/bin/
TOOLCHAIN=GCC_ARM
TARGET=CUSTOM_TARGET
```
Where `Nordic/toolchain/bin/arm-none-eabi-gcc --version`
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

cmonr, 2018-06-14:
Closing since issue appears to be resolved.

Feel free to open a new issue if this is not the case.

closed 2018-06-14 by cmonr