ARMmbed/mbed-os #4114: Automated test case "Example_2" is failed to compile.

pranavshandilya, 2017-04-05:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major
---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM, ARM & IAR

**DAPLink version:**
0243

**Expected behavior**
Test case should be compile and pass the test case.

**Actual behavior**
Failed to compile.

**Steps to reproduce**
Open Git Bash
Go to tools directory of Mbed OS 5.4.0 package.
Type below command
python singletest.py -i test_spec.json -M muts_all.json -n EXAMPLE_2
[image]

----------------------------------------------------------------

## Question
**How to resolve this issue?**

theotherjimmy, 2017-04-05:
@pranavshandilya Could you post the content of your test_spec.json and muts_all.json?

Could you also test with the latest master?

pranavshandilya, 2017-04-06:
Thanks for your quick reply, Below is the contents of muts_all.json & test_spec.json
muts_all.json
----
{
    "1" : {"mcu": "LPC1768",
        "port":"COM7",
        "disk":"F:\\",
        "peripherals": [""]
    }
}
----

test_spec.json
--
{
    "targets": {
        "LPC1768" : ["GCC_ARM", "ARM", "IAR"]
     }
}
----

theotherjimmy, 2017-04-06:
FYI @pranavshandilya You can surround your code blocks with ` ``` ` to make them format correctly. I edited your comment to add that formatting.

theotherjimmy, 2017-04-06:
@0xc0170 I'm not yet convinced that this is a tools issue.

theotherjimmy, 2017-04-06:
@bridadan I'm not as experienced with this part of our build flow. Could you take a look at this?

bridadan, 2017-04-06:
Looks like that test was removed when the filesystem changes went in. That test should have been removed from `tests.py`. Looking at `tests.py` now reveals quite a few test cases that are missing actual source files. I'll make a PR to remove them.

theotherjimmy, 2017-04-25:
@pranavshandilya That test was removed. Can we close this issue?

pranavshandilya, 2017-04-26:
Thanks @theotherjimmy

closed 2017-04-26 by pranavshandilya