ARMmbed/mbed-os #4116: NRF51_DK_BOOT completely different from NRF51_DK

PietKnaak, 2017-04-05:
We have an NRF51822 chip on which we want to enable FOTA. Code has been developed for the NRF51_DK and we try to switch to NRF51_DK_BOOT. 

There we run into compiler errors. Little research shows that NRF51_DK is re-written while NRF51_DK_BOOT is not, and missing a lot of functionality.

Can we easily make a new target base on NRF51_DK with FOTA support? What are  the risks? Or are we forced to downgrade our code to be compatible with NRF51_DK_BOOT (e.g. no spi async etc)?

nvlsianpu, 2017-04-06:
https://github.com/ARMmbed/mbed-os/blob/feature-nrf5_dfu_s13x_v2/targets/TARGET_NORDIC/README.md

nvlsianpu, 2017-04-06:
Unfortunately this feature is postponed to be merged into main sources. mbed-os is going to provide more generic DFU support than these on the feature branch for nrf5. @pan- Do you know whether this will be merged at all?

pan-, 2017-04-06:
@nvlsianpu It will be after the update of the `DFUService` class ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/ble/services/DFUService.h)).

pan-, 2020-10-06:
I'm closing this. There is no plan to update the NRF51 target on 5.15 (it has been removed from mbed os 6).

closed 2020-10-06 by pan-