ARMmbed/mbed-os #4117: Networking: sockets

chrissnow, 2017-04-05:
### Description
- Type: Bug

- Priority: Major 
---------------------------------------------------------------
## Bug

**Target**
LPC1768 (most likely actually LWIP rather than an NXP thing)

**Toolchain:**
ARM

Providing an IP instead of a DNS name according the header should be acceptable but this does not seem to work, a DNS name will resolve and work fine.

```c++
 /** Send a packet over a UDP socket
     *
     *  Sends data to the specified address specified by either a domain name
     *  or an IP address and port. Returns the number of bytes sent from the
     *  buffer.
     *
     *  By default, sendto blocks until data is sent. If socket is set to
     *  non-blocking or times out, NSAPI_ERROR_WOULD_BLOCK is returned
     *  immediately.
     *
     *  @param host     Hostname of the remote host
     *  @param port     Port of the remote host
     *  @param data     Buffer of data to send to the host
     *  @param size     Size of the buffer in bytes
     *  @return         Number of sent bytes on success, negative error
     *                  code on failure
     */
    nsapi_size_or_error_t sendto(const char *host, uint16_t port,
            const void *data, nsapi_size_t size);
```

theotherjimmy, 2017-04-05:
Hi @chrissnow. Thanks for the issue report. Could you change the name of this issue to reflect the issue at hand? "Networking: sockets" is pretty vague.

theotherjimmy, 2017-04-05:
Maybe something like `LWIP: allow sendto() to accep IP addresses` would be much more descriptive, supposing that it's accurate.

chrissnow, 2017-04-06:
This might actually may be fine and the issue is somewhere else in my code as I cant reliably recreate it.

chrissnow, 2017-04-06:
so this has gotten a bit strange!
it seems to break if the IP has a zero in it.
for example 192.168.0.46 will not work but 192.168.1.46 will.
a DNS name that resolves to an address with a zero does however work!

chrissnow, 2017-04-07:
@geky  is this something you may be able to help me with?

chrissnow, 2017-04-12:
I'm going to mark this closed as it's not as I originally described.

the real problem seems to be LWIP needs more memory if you try to connect to an IP on the same subnet than a routed IP or DNS name, increasing the thread size resolved the problem.

closed 2017-04-12 by chrissnow

geky, 2017-04-12:
Sorry I wasn't able to get to this earlier, but thanks for updating the issue 👍 

Was it the main stack that needed increasing?

chrissnow, 2017-04-13:
@geky I have a thread that dealt with maintaining the ethernet connection and a TCP client socket, I assigned the thread a 1024 stack and this works fine unless you connect to a link local address, 2048 makes it work.

Luckily I have some spare RAM, but otherwise a really odd bug!