ARMmbed/mbed-os #4120: MAX32625MBED devices enumerate with same target ID .

studavekar, 2017-04-05:
### Description
- Type: Bug 

MAX32625MBED devices enumerate with same target ID , seems to cause duplicate comm port. This is blocking CI.
```
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| MAX32625MBED  | MAX32625MBED[0]      | D:          | COM11       | 04150000b08042ed00000000000000000000000097969904 | 0242            |
| MAX32625MBED  | MAX32625MBED[1]      | E:          | COM11       | 04150000b08042ed00000000000000000000000097969904 | 0242            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```
---------------------------------------------------------------
## Bug

**Target**
MAX32625MBED  

**Toolchain:**
None

**Toolchain version:**
None
**mbed-cli version:** 
None

**meed-os sha:**
None

**DAPLink version:**

DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 04150000b08042ed00000000000000000000000097969904
HIC ID: 97969904
Auto Reset: 0
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0242
Bootloader Version: 0242
Git SHA: 30ebdca8c6a1e044708b410b9beb99ba1a92cca5
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Bootloader CRC: 0xa9f75799
Interface CRC: 0x3d118a82

**Expected behavior**
It should have unique target ID, and Unique comm port.

**Actual behavior**
It has duplicate target ID, and duplicate comm port.

**Steps to reproduce**

1) Connect 2 MAX32625MBED devices 
2) run mbedls

**Suggestion **

Probably needs to implement 

https://github.com/mbedmicro/DAPLink/

void read_unique_id(uint32_t *id);
void create_unique_id(void);

----------------------------------------------------------------

0xc0170, 2017-04-06:
cc @kgills @jeremybrodt

kgills, 2017-04-21:
I know what the problem is, just haven't had time to address it yet.

kgills, 2017-05-04:
Would it be acceptable to provide two separate firmware images for the CI devices? I'm having trouble generating unique target IDs on the devices.

studavekar, 2017-05-04:
@kgills Below are the requirement of CI system 

-  Device must enumerate on different com port for testing.

-  Device must have unique target_id for able to test on device selectivity. 

If above cases are satisfied we should be good, However it would be good to have unique target_id for all device.

studavekar, 2017-10-02:
@maclobdell @bentcooke this is blocking CI , we can't connect 2 maxim device on same machine.

This would mean we may have to stop support for maxim till this issue is fixed.

maclobdell, 2017-10-09:
@jessexm - can you help with this?

bentcooke, 2018-06-18:
This has been fixed.  Here is mbedls output showing different target_ids for a recent batch of boards:
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                        | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| MAX32625MBED  | MAX32625MBED[0]      | /mnt/pci-0000_00_1a_0-usb-0_1_4_4_1_0-scsi-0_0_0_0 | /dev/ttyACM0 | 04150000a453075000000000000000000000000097969904 | 0242            |
| MAX32625MBED  | MAX32625MBED[1]      | /mnt/pci-0000_00_1a_0-usb-0_1_4_1_1_0-scsi-0_0_0_0 | /dev/ttyACM1 | 041500009100070a00000000000000000000000097969904 | 0242            |
| MAX32625PICO  | MAX32625PICO[0]      | /mnt/pci-0000_00_1a_0-usb-0_1_4_2_1_0-scsi-0_0_0_0 | /dev/ttyACM3 | 041600001edc00ce00000000000000000000000097969904 | 0243            |
+---------------+----------------------+----------------------------------------------------+--------------+--------------------------------------------------+-----------------+

closed 2018-06-28 by 0xc0170