ARMmbed/mbed-os #4124: BLE can not work on MAX32600BED+EM9301

ExpandOcean, 2017-04-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

I am using MAX32600MBED to develop BLE app for max32600 devices. The BLE chip is EM9301. I found the board MAXWSNENV supports this chip and is also max32600 device. So I thought it should be easy to use MAXWSNENV code to run BLE stack. It's very weird that I can not find schematic of MAXWSNENV, what I can find is only a manual of this board but it does not say anything useful about the BLE chip(not even the name/id of the BLE chip). I checked the MAXWSNENV's code(https://github.com/ARMmbed/mbed-os/commit/d5351807844b77586f3b53f66f046b66d64f445c) , and found that the SPI/RST/IRQ pins in MaximBLE.cpp were defined in PinNames.h. So I just replace the HCI pins with the actual pins I used(P2_0~P2_5), but it looks can not work. It hanged on "ble.init(bleInitComplete);". Then I found that even if I deleted the line 75 of MaximBLE.cpp(static SPI _spi(HCI_MOSI, HCI_MISO, HCI_SCK, HCI_CSN);), the code can still be compiled. I am confused, why the _spi is not used. If it is no use, where should I define the acture SPI.

**Target**
MAX32600

**Toolchain:**
GCC_ARM

0xc0170, 2017-04-06:
cc @kgills @jeremybrodt

cmonr, 2018-01-23:
Closing since support is currently present in mbed-os.

closed 2018-01-23 by cmonr