ARMmbed/mbed-os #4130: Test fails if device is full without meaningful error

kvanberendonck, 2017-04-07:
Details are in this thread:
https://developer.mbed.org/questions/73957/Changing-name-of-test-affects-whether-it/?c=27025

Over time, the device fills up with useless bin files and you start getting a non helpful error. If tests fail due to no space on device, it should print an error telling you to make more space.

0xc0170, 2017-04-10:
cc @theotherjimmy @bridadan

bridadan, 2017-04-10:
@kvanberendonck Thanks for raising the issue. This is unfortunately a limitation of that particular mbed's firmware (it is one of about 3 boards that have this interface, so it is not the "norm"). It's actually a side effect of the "LocalFilesystem" feature. It lets you have a persistent filesystem, but as you have observed, its not the greatest for automated testing. And I agree that the error should be better.

We have encountered this problem before. With the mbed 2 test system, you could specify the name of the binary to flash on the device, so we always made sure they had the same name and it overwrote the previous binary. Perhaps we could do something similar in greentea/htrun/mbed test?

Are you planning on using this platform for automated testing? The changes would be relatively simple in the tools, I'd be happy to point you in the right direction if you had any interest in contributing. That would probably be the fastest way this would get fixed. Otherwise it'd get entered into our backlog and we would get to it at some point in the future. Though I have a feeling it wouldn't be a high priority since this board is generally the exception when it comes to interface firmware behavior.

cmonr, 2018-01-23:
Closing due to inactivity. I don't think this is still an issue, but if it is, feel free to reopen it for re-investigation.

closed 2018-01-23 by cmonr

theotherjimmy, 2018-01-23:
@donkeybonks This issue is with the board's DAP firmware, and we can't fix that. The interface chip predates the ARM Cortex-M line of processors, and is not supported by DAPLink.