ARMmbed/mbed-os #4138: Unable to check the build details

nikitateggi, 2017-04-08:
after compiling any code in the mbed compiler, I am unable to see the build details. i.e. flash and RAM size. Please address this issue.

pranavshandilya, 2017-04-08:
@nikitateggi please follow the below steps to see the build details and for attached snapshot is for your reference:-
Step 1. Compile any code (e.g. mbed-os-example-blinky for LPC1768) 

[image]

Step 2. Click on "Build Details".

[image]

Ste 3. Now you can see the build details i.e. Flash and RAM size.

[image]

If this is your issue then OK otherwise please give me more clarification of your issue.

Thanks

nikitateggi, 2017-04-10:
yes. This is the issue.

pranavshandilya, 2017-04-10:
@nikitateggi you have to use an offline compiler to see the build details ( flash and RAM size).
Export your program and generate the map file, where you can find the details about Memory usage.
Steps for exporting the program: 
1. Right click on your program
             - And click on fifth option " Export Program"
2. Choose your Toolchain (IDE)
             - Like Keil, IAR, CooCox etc.
3. Compile the program using offline compiler
             - After compilation, It'll create a BUILD folder
               Inside the BUILD folder, Search for map(Extension :- *.map) file.
4. Open the map file using notepad.
    Goto end of the map file there you can see the memory usage.

0xc0170, 2017-04-10:
@nikitateggi For future reference, please provide more details when describing an issue (any detail that would help us to reproduce the issue easily). Thanks

nikitateggi, 2017-04-22:
@pranavshandilya Thank you. But why does the mbed doen't show the flash and ram memory?

nikitateggi, 2017-04-22:
@0xc0170 Sure.

pranavshandilya, 2017-04-22:
Welcome, @nikitateggi

`But why does the mbed doesn't show the flash and ram memory?`
I'm not a frequent user of the online compiler, that's why I didn't notice these things.
So, sorry for that.

@0xc0170  Please, Can you help?

0xc0170, 2017-04-24:
cc @theotherjimmy

theotherjimmy, 2017-04-24:
duplicate of #3967

theotherjimmy, 2017-04-24:
Please continue discussion in #3967

closed 2017-04-24 by theotherjimmy