ARMmbed/mbed-os #4139: Compilation warning related to mbed_board.c and mbed_retarget.cpp files.

pranavshandilya, 2017-04-08:
### Description
- Type: Enhancement
- Priority: Minor
---------------------------------------------------------------
**Target**
LPC1768

**Toolchain:**
GCC_ARM | ARM | IAR

**Expected behavior**
Should we compile without any warning.

**Actual behavior**
At the time of compiling, It's giving warnings.

---------------------------------------------------------------
## Question
When I'm building the code for LPC1768, below warning is showing:-
Toolchain: - GCC_ARM
 mbed_board.c@99,36: comparison between signed and unsigned integer expressions [-Wsign-compare]
Why this warning is not showing when I build with different toolchain(ARM & IAR)?
------snip------
    int size = vsprintf(buffer, format, arg);
    if (size > 0) {
        if (!stdio_uart_inited) {
            serial_init(&stdio_uart, STDIO_UART_TX, STDIO_UART_RX);
        }
#if MBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES
        for (unsigned int i = 0; i < size; i++) {
            if (buffer[i] == '\n' && stdio_out_prev != '\r') {
                 serial_putc(&stdio_uart, '\r');
            }
------snip------
In the above code, 'size' is the type of signed integer and 'i' is the type of unsigned integer
So, this comparison will give a warning but when I build with different toolchain(ARM & IAR),
this warning is not there..why?

Toolchain: - ARM & IAR
mbed_retarget.cpp@123,0:  #177-D: function "openmode_to_posix" was declared but never referenced
Why this warning is not showing with GCC_ARM?

Please address this issue.

0xc0170, 2017-04-10:
will be fixed via https://github.com/ARMmbed/mbed-os/pull/4141

thanks for reporting

pranavshandilya, 2017-04-19:
Thanks @0xc0170

closed 2017-04-19 by pranavshandilya