ARMmbed/mbed-os #4146: CMSIS5: NRF52840_DK needs updating to RTX2

bulislaw, 2017-04-10:
The NRF52840_DK target fails to build using `feature_cmsis5` branch (which updates CMSIS and RTX to versions 5 and 2), the two issues:
* NVIC set/get vector are defined for the target (CMSIS already defines them)
* targets/TARGET_NORDIC/TARGET_NRF5_SDK13/sdk/libraries/timer/app_timer_rtx.c file is braking RTX abstraction layers using it's internals and needs to be ported to RTX2 (or fixed if possible)

CC: @pan- @nvlsianpu

nvlsianpu, 2017-04-12:
1) This is similar situation as was for NRF5 ports. NVIC set/get vector must been implemented by this target.
I remember that i added "CMSIS_VECTAB_VIRTUAL" macro to nRF52832 MCU target description. Now I see that there is another one "CMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"".

2) #4175 removes unwanted app_timer_rtx.c file.

bulislaw, 2017-04-13:
Great, thanks. I'll deal with 1.

closed 2017-04-19 by bulislaw