ARMmbed/mbed-os #4151: SDFileSystem example broken on mbed 2 library rev 138 or onward

toyowata, 2017-04-11:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/issues/3983
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F, SSCI824, GR-PEACH

**Toolchain:**
ARM

**Toolchain version:**
armcc 5.06 (online compiler)

**Expected behavior (Rev. 137 or older)**
SDFileSystem_HelloWorld example works with log below:
```
Hello World!
Goodbye World!
```

**Actual behavior (Rev. 138 or onward)**
SDFileSystem_HelloWorld example fails with log below:
```
Hello World!
Could not open file for write
```
**Steps to reproduce**
* Import following example with check ```Update all libraries to the latest revision``` option.
https://developer.mbed.org/users/mbed_official/code/SDFileSystem_HelloWorld/
* Modify the definition of the SDFileSystem constructor below:
```
#if defined(TARGET_LPC1768)
SDFileSystem sd(p5, p6, p7, p8, "sd"); // the pinout on the mbed Cool Components workshop board
#elif defined(TARGET_SSCI824)
SDFileSystem sd(dp8, dp9, dp10, dp11, "sd");
#else // CI test shield
SDFileSystem sd(D11, D12, D13, D10, "sd");
#endif
```
* Compile and run

**Notes**
* The mbed library rev 137 or older revision works fine
* The mbed library rev 138 or onward generate warning messages in compile below:
```
Warning: Function "mbed::DirHandle::DirHandle()" (declared at <a href="#" onmousedown="mbed_doc_goto('/SDFileSystem_HelloWorld_Update//extras/mbed_093f2bd7b9eb/drivers/DirHandle.h', '48'); return false;">/extras/mbed_093f2bd7b9eb/drivers/DirHandle.h:48</a>) was declared "deprecated" in "SDFileSystem/FATFileSystem/FATDirHandle.cpp", Line: 28, Col: 55
Warning: Function "mbed::FileHandle::FileHandle()" (declared at <a href="#" onmousedown="mbed_doc_goto('/SDFileSystem_HelloWorld_Update//extras/mbed_093f2bd7b9eb/drivers/FileHandle.h', '45'); return false;">/extras/mbed_093f2bd7b9eb/drivers/FileHandle.h:45</a>) was declared "deprecated" in "SDFileSystem/FATFileSystem/FATFileHandle.cpp", Line: 28, Col: 39
Warning: Function "mbed::FileSystemLike::FileSystemLike(const char *)" (declared at <a href="#" onmousedown="mbed_doc_goto('/SDFileSystem_HelloWorld_Update//extras/mbed_093f2bd7b9eb/drivers/FileSystemLike.h', '47'); return false;">/extras/mbed_093f2bd7b9eb/drivers/FileSystemLike.h:47</a>) was declared "deprecated" in "SDFileSystem/FATFileSystem/FATFileSystem.cpp", Line: 45, Col: 62
Success: Success! in "<a href="#" onClick="mbed_navigate_to('/SDFileSystem_HelloWorld_Update', function() { theitf.controls.get('sidebar_tabs').activate(1); }); return false;">Build Details</a>"
```

0xc0170, 2017-04-12:
cc @geky

ElectronicaXAB3, 2017-07-07:
Will this bug be solved? Because in revision 146, the bug is not allowing a compile to run.

toyowata, 2017-07-07:
In revision 146:
```
Warning: Overloaded function "mbed::FileSystemHandle::open" is hidden by "FATFileSystem::open" -- virtual function override intended? in "SDFileSystem/FATFileSystem/FATFileSystem.h", Line: 48, Col: 26
Warn
ing: Overloaded function "mbed::FileSystemHandle::open" is hidden by "FATFileSystem::open" -- virtual function override intended? in "SDFileSystem/FATFileSystem/FATFileSystem.h", Line: 48, Col: 26
Error: Object of abstract class type "FATFileHandle" is not allowed in "SDFileSystem/FATFileSystem/FATFileSystem.cpp", Line: 100, Col: 17
Error: Object of abstract class type "FATDirHandle" is not allowed in "SDFileSystem/FATFileSystem/FATFileSystem.cpp", Line: 136, Col: 17
Info: Unable to download. Fix the reported errors...
```

cc @0xc0170 @geky

geky, 2017-07-10:
Sorry, I didn't realize this wasn't fixed as a part of the LocalFileSystem fixes (https://github.com/ARMmbed/mbed-os/pull/4087). Since the old FATFileSystem is outside of the mbed OS repo, it needs to be updated seperately.

I'm not entirely sure the best way to update the old FATFileSystem, so I put up a pr here:
https://developer.mbed.org/teams/mbed-official/code/FATFileSystem/pull-request/4

0xc0170, 2017-07-11:
> I'm not entirely sure the best way to update the old FATFileSystem, so I put up a pr here:
https://developer.mbed.org/teams/mbed-official/code/FATFileSystem/pull-request/4

I assume this is backward compatible. @sg- We shall review this fix

0xc0170, 2017-07-20:
The above PR was merged, the latest filesystem with mbed-rtos should be all functional.
@toyowata I'll close this. Please reopen with update if needed

closed 2017-07-20 by 0xc0170