ARMmbed/mbed-os #4181: nRF51822 pstorage APIs are broken since mbed OS 5.2

janjongboom, 2017-04-12:
### Description
- Type: Bug
- Priority: Blocker

## Bug

**Target**
nRF51-DK, also seen on Minewtech nRF51 and Minewtech nRF52

**Toolchain:**
GCC_ARM and ARMCC

**Toolchain version:**
4.9.3 and online compiler

**meed-os sha:**
The problem is present in mbed OS 5.2.3 and 5.4.2, but not in mbed OS 5.1.5.

**Expected behavior**
Take https://github.com/roywant/EddystoneBeacon. Change the adv interval (3rd characteristic, has value 0x2bc), and persist. On mbed OS 5.1.5 this is persisted in persistent storage. mbed OS 5.2.3 does not persist.

The first call to `pstorage_store` goes fine, but then a call to `pstorage_update` will wipe out the flash... Just restart the board and inspect the persistence signature (not even required to power cut). I see that we used to get 2x `pstorageNotificationCallback` after updating, but now only one... Maybe that's a hint?

I bisected the problem:

```
e67117b8eafb6dada5f0cc4c09e140c95a001cb3 is the first bad commit
commit e67117b8eafb6dada5f0cc4c09e140c95a001cb3
Author: Andrzej Puzdrowski <andrzej.puzdrowski@nordicsemi.no>
Date:   Tue Aug 23 08:52:32 2016 +0200

    introduce the SDK nRF5 Peer manager into BLE security features

    PM will be used for SoftDevices s13x, otherwise DM will be used.

    Fix - security key distribution settings - Signing is not supported.

    declare usage of encryption LTK and IRK if boonding enabled

    deleted unused local reference to nRF5xGap class in bleGattcEventHandler.
    fix PM internal bug (For gcc with -0s optimization an application fail to save a boonding data)
    fuse redeclaration of PACKED macro in nRF5 SDK sources
```

Seems legit.

pan-, 2017-04-13:
@nvlsianpu Do you know if `fstorage` and `pstorage` can be used concurrently ? 
@janjongboom It might be interesting to try with `IS_LEGACY_DEVICE_MANAGER_ENABLED ` set to 1 ([here](https://github.com/ARMmbed/mbed-os/blob/e67117b8eafb6dada5f0cc4c09e140c95a001cb3/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF5/source/btle/btle.cpp#L36)).

nvlsianpu, 2017-04-25:
Thanks for the bug report.

`pstorage` and `fstorage` can't being used concurrently. `pstorage` is deprecated module, replaced by `fstorage`. Mentioned commit switched security manger to use fstorage by deafult.

0xc0170, 2018-06-19:
I believe this one can be closed. Resolved via referenced PR above  to the ble example.

closed 2018-06-19 by 0xc0170