ARMmbed/mbed-os #4182: Adding change directory ability to FileSystem

mbartling, 2017-04-12:
### Description
- Type: Enhancement
- Priority: Minor
----------------------------------------------------------------
## Enhancement
having to prepend "/sd/" or "/fs/" to every filename is tedious and can lead to a bunch of polluted code. It would be nice to have the ability to change directories into the mounted root.

for example:

```cpp
#include "mbed.h"
#include "FATFileSystem.h"
#include "HeapBlockDevice.h"
#include <stdio.h>
#include <errno.h>

HeapBlockDevice bd(128 * 512, 512);
FATFileSystem fs("fs");

...

  error = fs.mount(&bd, /*mount_as_root=*/ true);
  return_error(error);

  printf("Opening a new file, numbers.txt.");
  FILE* fd = fopen("numbers.txt", "w");
```

comments/concerns?

deepikabhavnani, 2018-09-11:
@mbartling  - Storage has changed quite a bit since this issue, closing it. Please see if `get_default` for fs / block device and partition will help with this or not.

closed 2018-09-11 by deepikabhavnani

geky, 2018-09-19:
We still can't reference filesystems as "root", which is what I believe the main feature request of this issue is. This may be fixed in future work around mount points.