ARMmbed/mbed-os #4188: Issue in Serial RX interrupt processing?

maiorfi-at-innovactive, 2017-04-14:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F410RB

**Toolchain:**
ARM

**Toolchain version:**
5.x

**Expected behavior**
led blinking when data received by serial RX channel

**Actual behavior**
program halts in HardFault_Handler when flushing serial backing stream

**Steps to reproduce**

```c++
#include "mbed.h"

DigitalOut led1(LED1);

Serial pc(USBTX, USBRX);

void callback_ex() {
    // Note: you need to actually read from the serial to clear the RX interrupt
    pc.getc();
    led1 = !led1;
}

int main() {
    pc.attach(&callback_ex);
}
```

maiorfi-at-innovactive, 2017-04-15:
It seems that using Serial instead of RawSerial with RTOS APIs and/or within rx interrupt handlers doesn't work as expected because of thread affinity issues in stdio stream backing Serial class.

Maybe kind of best-practice guidelines for properly implementing serial protocols management (such as AT command based ones) with RTOS apis would be useful.

AnotherButler, 2017-04-19:
@maiorfi-at-innovactive That's a great idea. We are currently working on a best-practices page. I'll be sure to make sure we include that. Thanks for the suggestion.

janjongboom, 2017-12-01:
@sg- Why is this tagged with 'type: enhancement'? If a board crashes when attaching serial interrupt that's a bug, not an enhancement.

Also reported on Stack Overflow: https://stackoverflow.com/questions/47495759/stm32f410rb-nucleo-not-reading-from-serial-device

closed 2018-12-19 by ciarmcom