ARMmbed/mbed-os #4189: STM32s Serial does not properly handle parity bits.

Sissors, 2017-04-15:
### Description
- Type: Bug
- Priority: Intermediate

---------------------------------------------------------------
## Bug

**Target**
STM32s, haven't checked if all suffer from it.

**Toolchain:**
All

**Expected behavior**
Enabling parity bit adds a parity bit to the transmission.

**Actual behavior**
Enabling parity bit replaces a bit for the parity bit in the transmission. If you look at the serial format code, https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F0/serial_api.c#L300, you see 9-bit transmission is only enabled if the format is 9 data bits. However according to the reference manual, the number of data bits is including the parity bit: So if you have it set on 8-bits with parity, it will send 7 data bits and the parity. So if you want 8 data bits and parity, you need to set it on 9-bit data. If you call the mbed serial format function with 9 data bits and parity, it indeed correctly sends 8 data bits and parity. But obviously you should be able to set the format to 8 data bits and have a proper output format.

Kudos to: https://gathering.tweakers.net/forum/list_message/50904807#50904807 for immediatly knowing what the issue was. 

**Steps to reproduce**
Send a serial packet with parity enabled and check with logic analyzer/serial terminal on PC.

fmanno, 2017-04-22:
I agree with @Sissors that the way of specifying the configuration is confusing. Serial applications usually specify configuration in the format <data_length - parity - stop_bits> e.g. 8-N-1, 7-E-2, etc. I.e. the parity bit is excluded from the data_bits count. However from the reference manuals of both  [STM32F0](http://www.st.com/content/ccc/resource/technical/document/reference_manual/c2/f8/8a/f2/18/e6/43/96/DM00031936.pdf/files/DM00031936.pdf/jcr:content/translations/en.DM00031936.pdf) and [STM32F4](http://www.st.com/content/ccc/resource/technical/document/reference_manual/5d/b1/ef/b2/a1/66/40/80/DM00096844.pdf/files/DM00096844.pdf/jcr:content/translations/en.DM00096844.pdf) It says that when parity is enabled, it replaces the MSB of the data_length. Thus implying that with parity disabled there's actually up to 9 bits usable length for the serial data transmissions. In those cases always reserving one bit for the parity in the format say 8-N-1 doesn't seem efficient as some applications may benefit from the extra bit, who knows.

Having said that, it's my interpretation that mbedOS aims mainly at portability between different chips and boards and keeping that in mind I've been looking at the [MAX32625](https://pdfserv.maximintegrated.com/en/an/AN6350.pdf) and the [SAMG55](http://www.atmel.com/Images/Atmel-11289-32-bit-Cortex-M4-Microcontroller-SAM-G55_Datasheet.pdf) user manuals and in both cases the data_length excludes the parity bit (i.e. they stick to the 8-N-1 style configuration). So I believe that the serial_api.c implementation for the STM32 family should also follow that way of specifying the format or users might get a nasty surprise when running their app in an STM32 platform if they used another one before.

I've created [Pull request 4216](https://github.com/ARMmbed/mbed-os/pull/4216) with a change that addresses this.

0xc0170, 2017-04-24:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-04-24:
@fmanno - thank you for the PR - this looks good to me

PR #4216 (2017-06-07): STM32s Serial: Correct handling of parity bits
## Description
Reworked the serial_format() function for STM32F0x devices to take the format in the form:
`data_bits - parity - stop_bits`
(E.g. 8 - N - 1) where data_bits exclude the parity bit.
Added a case for 7 bits data as at least STM32F0x1/STM32F0x2/STM32F0x8 support 7 bits data.
Consolidated serial_format() and uart_init() functions into a general TARGET_STM serial_api.c file since the functions are common to all STM targets.

## Status
DONE

## Migrations
YES
After this is merged, behaviour for STM32s will be consistent with the API behav [... cut]
commit 5e98da14 (2017-06-07): STM32s Serial does not properly handle parity bits

Reworked the serial_format() function for STM32F0x
devices to take the format in the form:
data_bits - parity - stop_bits

E.g. 8 - N - 1

where data_bits exclude the parity bit.
Added a case for 7 bits data as at least the chips
STM32F0x1/STM32F0x2/STM32F0x8 support 7 bits data.

Consolidated serial_format() and uart_init()
functions into a general TARGET_STM serial_api.c
file since the functions are common to all STM targets.

Fixes #4189
closed 2017-06-07 by sg-