ARMmbed/mbed-os #4191: Deepsleep and sleep does not work in STM32L0 series

rbonghi, 2017-04-16:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32L0 series

**Toolchain:**
ARM

**Toolchain version:**
5.x

**Expected behavior**
Go to sleep or deepsleep

**Actual behavior**
Skip the functions sleep() and deepsleep()

**Steps to reproduce**
``` c
int main()
{ 
    while (1) {
            printf("Entering sleep\n");
            sleep();
            printf("Entering deepsleep\n");
            deepsleep();
        }
    }
}
```

----------------------------------------------------------------
## Enhancement

**Suggested enhancement**
From release 137 https://developer.mbed.org/users/mbed_official/code/mbed/rev/e1686b8d5b90 the functions deepsleep() and sleep() does not work. If w go to release 136 the deepsleep and sleep functions works. I Think the error is from 137 release.

joaogmauricio, 2017-04-18:
Hi Raffaello,

I shall add that the issue also affects Nucleo L476LG boards.

Best regards,
João

bulislaw, 2017-04-18:
Sleep was disabled in develop (default) build profile https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/ it's going to change soon
https://github.com/ARMmbed/mbed-os/pull/4097

bulislaw, 2017-04-25:
`sleep()`/`deepsleep()` should work for develop and release profiles on master, for platforms it's defined for. I'll close this issue, please reopen it if you have any further questions.

closed 2017-04-25 by bulislaw