ARMmbed/mbed-os #4193: mbed os 2.0, bind in UDPSocket

sstaub, 2017-04-18:
Can you extend the UDPSocket API with an **int bind(const char *address, int port)** function in mbed os 2.0?
Maybe that help me with a multicast problem described in #2766

0xc0170, 2017-04-18:
cc @geky @kjbracey-arm

geky, 2017-04-18:
UDPSocket has one:
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/Socket.h#L75-L84
``` cpp
/** Bind a specific address to a socket
 *
 *  Binding a socket specifies the address and port on which to recieve
 *  data. If the IP address is zeroed, only the port is bound.
 *
 *  @param address  Null-terminated local address to bind
 *  @param port     Local port to bind
 *  @return         0 on success, negative error code on failure.
 */
nsapi_error_t bind(const char *address, uint16_t port);
```

It's defined in `Socket`, the common base class for `TCPSocket` and `UDPSocket`.

Unfortunately, I believe IPv4 multicast is still a work in progress (https://github.com/ARMmbed/mbed-os/issues/3512, https://github.com/ARMmbed/mbed-os/issues/3512)

sstaub, 2017-04-18:
@geky you are right on mbed os5 but not in os2 that is here [https://github.com/ARMmbed/mbed-os/blob/master/features/unsupported/net/lwip/Socket/UDPSocket.h](url)

geky, 2017-04-18:
Ah, looks like your right. It may be better to focus effort on fixing the missing features in the most current socket api.

SeppoTakalo, 2018-06-27:
`features/unsupported` means it is unsupported.
So no fixes, no new functionalities.

closed 2018-06-27 by 0xc0170