ARMmbed/mbed-os #4194: mbed os should not expose device unique IDs via MAC address

bremoran, 2017-04-18:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**mbed-os sha:**
65adf446c52fa4e3045d12c761f23948d0d271fd

**Expected behavior**
mbed OS should use a non-reversible hashing function to transform any on-device unique ID

**Actual behavior**
mbed OS uses CRC16 to transform the unique ID into a MAC address.

Users of mbed OS may consider the on-chip unique ID to have sufficient entropy to act as a device-private secret. If mbed OS exposes the unique ID in any way without transforming it with a strong hashing function, it could expose some portion of the device-private secret.

The current implementation of `mbed_mac_address()` leaks information about the unique ID in a way that a use may not expect:

https://github.com/ARMmbed/mbed-os/blob/920db8e4a33fb5c13ccda5f9a39d6434562bc956/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed_overrides.c#L57-L59

This could reduce the potential entropy of a device-private secret from 128 bits to 80 bits. If the MAC address were calculated with a secure hash, such as SHA-256, this would not be a problem.

cc @bulislaw @marcuschangarm

bulislaw, 2017-04-18:
I wonder how generic this issue is, I can see it tying up with our HAL work:
- unique ID API as part of HAL (access: direct? transformed?)
- MAC generation in target independent way?

@sg- @0xc0170

sg-, 2017-04-18:
Seems like a good idea. I'd suggest:
- UUID is direct from the device in the HAL. It can later be transformed further up the stack.
- should be many MAC implementations, generic, device specific, external IC/flash memory location provided

bremoran, 2017-04-19:
@sg- I agree. There need to be more choices for MAC assignment. Possibly including a purely random one (TRNG).

With regards to the UUID, however, can I suggest two guidelines?

* Any exposure to application code should be transparent: it shouldn't transform the UUID in any way.
* Any use of UUID that is not exposed to the application code (e.g. MAC address generation) should take care not to leak information, since this could compromise assumptions made by the developer.