ARMmbed/mbed-os #4195: Development board LPC1768

pranavshandilya, 2017-04-18:
### Description
- Type: Question
- Priority: Minor
---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM | ARM | IAR

**DAPLink version:**

**Expected behavior**
LED should be blink.

**Actual behavior**
LED is not blinking.

**Steps to reproduce**
1. Open online compiler.
2. Select Target LPC1768
3. Open main.cpp file in Mbed-OS blinky example code.
4. Change LED pin no. P2_3 (In my development board(LPC1768) LED is connected to P2_3)
5. Compile the code and copy binary file to drive(It's an ISP-supported development board).
----------------------------------------------------------------
## Question
After copying binary file to USB boot drive.
LED is not blinking.

Is it compulsory to use mbed enable board for online compiler or Mbed SDK?
If it's not then, **How to use?**

wdwalker, 2017-04-18:
Which LPC1768 development board do you have? The one I have (NXP's ARM mbed LPC1768) shows LED1 connected to P1_18: http://www.nxp.com/documents/design_support/ARM_mbed_LPC1768_Schematic.pdf, which is in line with the definitions in PinNames.h:

    LED1 = P1_18,
    LED2 = P1_20,
    LED3 = P1_21,
    LED4 = P1_23,

Will

pranavshandilya, 2017-04-19:
Explore Cortex-M3 LPC1768 (Board name)
http://exploreembedded.com/wiki/images/0/0e/LPC1768_Stick_Schematic.pdf
LED1 = P2_0,
LED2 = P2_1,
LED3 = P2_2,
LED4 = P2_3,
>which is in line with the definitions in PinNames.h:
I'm not passing LED name from application code when I'm creating object for 
DigitalOut class, Instead of LED name i'm passing Port name and pin.
Code is below.
snip---
**DigitalOut led(P2_3);**
int main(void)
{
    while(1) {
        led !=led;
        wait(0.5);
    }
}
------

toyowata, 2017-04-19:
@pranavshandilya 

>Compile the code and copy binary file to drive(It's an ISP-supported development board).

The binary which created by online compiler, does not have checksum code.  The mbed interface firmware of NXP LPC1768 platform adds checksum when the binary is flashed by USB drag & drop for you.  But in your case, you have your own board and does not have such feature.  So, you have to create correct vector table checksum by yourself using checksum utility.  Please see more detail below:

https://www.lpcware.com/content/forum/download-program-without-ide-#comment-1100233

pranavshandilya, 2017-04-19:
Thanks for your support @toyowata

closed 2017-04-19 by pranavshandilya