ARMmbed/mbed-os #4196: greentea test : race_test is failing in Cortex-M3(LPC1347)

pranavshandilya, 2017-04-19:
### Description
- Type: Bug
- Related issue: [https://github.com/ARMmbed/mbed-os/issues/3273](url)
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LPC1347

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release)

**mbed-cli version:** 
1.0.0

**DAPLink version:**
0243

**Expected behavior**
A test case should be a pass.

**Actual behavior**
Test case is failing.

**Steps to reproduce**
1. Download mbed-os-5.4.0
2. Go to the root directory.
3. open targets.json file mbed-os-mbed-os-5.4.0/targets/targets.json
4. Add extra macros : `"default_lib": "small",`
4. Run the below command
5. mbed test -m LPC1347 -t GCC_ARM -n tests-mbed_drivers-race_test -v

```
mbedgt: mbed-host-test-runner: started
[1492584165.61][HTST][INF] host test executor ver. 1.1.7
[1492584165.61][HTST][INF] copy image onto target...
[1492584165.61][COPY][INF] Waiting up to 60 sec for '701100000000000000000000000000000000000097969904' mount point (current is 'F:')...
        1 file(s) copied.
[1492584210.06][HTST][INF] starting host test process...
[1492584210.62][CONN][INF] starting connection process...
[1492584210.62][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1492584210.62][CONN][INF] initializing serial port listener...
[1492584210.62][PLGN][INF] Waiting up to 60 sec for '701100000000000000000000000000000000000097969904' serial port (current is 'COM7')...
[1492584210.65][HTST][INF] setting timeout to: 60 sec
[1492584251.30][SERI][INF] serial(port=COM7, baudrate=9600, timeout=0.01)
[1492584251.30][SERI][INF] reset device using 'default' plugin...
[1492584251.55][SERI][INF] waiting 1.00 sec after reset
[1492584252.55][SERI][INF] wait for it...
[1492584252.55][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1492584252.55][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1492584252.55][CONN][INF] sending preamble '87270836-ca0b-4286-86d3-88a0b5206a9c'
[1492584252.55][SERI][TXD] {{__sync;87270836-ca0b-4286-86d3-88a0b5206a9c}}
[1492584252.69][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1492584252.74][CONN][INF] found SYNC in stream: {{__sync;87270836-ca0b-4286-86d3-88a0b5206a9c}} it is #0 sent, queued...
[1492584252.75][HTST][INF] sync KV found, uuid=87270836-ca0b-4286-86d3-88a0b5206a9c, timestamp=1492584252.738000
[1492584252.75][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1492584252.75][HTST][INF] DUT greentea-client version: 1.3.0
[1492584252.79][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1492584252.80][HTST][INF] setting timeout to: 20 sec
[1492584252.82][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1492584252.83][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1492584252.83][HTST][INF] host test setup() call...
[1492584252.83][HTST][INF] CALLBACKs updated
[1492584252.83][HTST][INF] host test detected: default_auto
[1492584252.85][CONN][INF] found KV pair in stream: {{__testcase_count;2}}, queued...
[1492584252.86][CONN][RXD] >>> Running 2 test cases...
[1492584252.91][CONN][INF] found KV pair in stream: {{__testcase_name;function init race}}, queued...
[1492584252.96][CONN][RXD]
[1492584252.96][CONN][INF] found KV pair in stream: {{__testcase_name;class init race}}, queued...
[1492584253.00][CONN][RXD] >>> Running case #1: 'function init race'...
[1492584253.05][CONN][INF] found KV pair in stream: {{__testcase_start;function init race}}, queued...
[1492584253.07][CONN][RXD] Running function race test
[1492584253.35][CONN][RXD] :86::FAIL: Expected 1 Was 0
[1492584253.40][CONN][INF] found KV pair in stream: {{__testcase_finish;function init race;0;1}}, queued...
[1492584253.47][CONN][RXD] >>> 'function init race': 0 passed, 1 failed with reason 'Assertion Failed'
[1492584253.47][CONN][RXD]
[1492584253.55][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1492584253.57][CONN][RXD] >>> TESTS FAILED!
[1492584253.60][CONN][INF] found KV pair in stream: {{__testcase_summary;0;1}}, queued...
[1492584253.62][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1492584253.63][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1492584253.615000
[1492584253.65][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1492584253.65][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1492584253.66][HTST][INF] __exit(0)
[1492584253.66][HTST][INF] __notify_complete(False)
[1492584253.66][HTST][INF] __exit_event_queue received
[1492584253.66][HTST][INF] test suite run finished after 0.86 sec...
[1492584253.67][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1492584253.68][HTST][INF] CONN exited with code: 0
[1492584253.68][HTST][INF] No events in queue
[1492584253.68][HTST][INF] stopped consuming events
[1492584253.68][HTST][INF] host test result() call skipped, received: False
[1492584253.68][HTST][INF] calling blocking teardown()
[1492584253.68][HTST][INF] teardown() finished
[1492584253.68][HTST][INF] {{result;failure}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'FAIL'
mbedgt: test on hardware with target id: 701100000000000000000000000000000000000097969904
mbedgt: test suite 'tests-mbed_drivers-race_test' .................................................... FAIL in 88.73 sec
        test case: 'function init race' .............................................................. FAIL in 0.35 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.9083938845
mbedgt: test suite report:
```
 ----------------------------------------------------------------
## Question
I am unable to find the issue.
Could you please suggested some reference like this issue is related to RTOS or driver.

0xc0170, 2017-04-19:
Hi,

why are you setting default_lib to small? This changes the toolchain configuration to:

```
        if default_lib == "small":
            self.flags["common"].append("-DMBED_RTOS_SINGLE_THREAD")
            self.flags["ld"].append("--specs=nano.specs")

``

I would assume race test to display an error if ``MBED_RTOS_SINGLE_THREAD`` is defined ?

pranavshandilya, 2017-04-19:
@0xc0170  
Thank you for your comment.

`why are you setting default_lib to small?`
Because I am using 50% RAM of a controller and 50% is for some other purpose.
But I changed default_lib to std. the result is same.

`I would assume race test to display an error if ``MBED_RTOS_SINGLE_THREAD`` is defined ?`
It is not showing any error. The test case is failed. 

thanks

0xc0170, 2017-04-19:
It's expected failure. I can add there a warning, as might not be obvious when running tests. I'll propose this via a new PR to address this issue

0xc0170, 2017-04-20:
PR with a proposed addition sent

PR #4205 (2017-04-24): tests: race test - add not supported for single threaded env
Fixes #4196. As someone might not be aware that settting default_lib to small has
some implications regarding thread safety, therefore we print an error.
commit 84eddd01 (2017-04-24): tests: race test - add not supported for single threaded env

Fixes #4196. As someone might not be aware that settting default_lib to small has
some implications regarding thread safety, therefore we print an error.
commit ccc2e4b9 (2017-05-05): tests: race test - add not supported for single threaded env

Fixes #4196. As someone might not be aware that settting default_lib to small has
some implications regarding thread safety, therefore we print an error.
commit 3f549ad1 (2017-05-05): tests: race test - add not supported for single threaded env

Fixes #4196. As someone might not be aware that settting default_lib to small has
some implications regarding thread safety, therefore we print an error.
closed 2017-04-24 by theotherjimmy