ARMmbed/mbed-os #4213: Build tools leak compiler process in CI

mazimkhan, 2017-04-21:
Please see the screenshots:
[image]
[image]

Suspended/Zombi arm-none-eabi-g++.exe prevent recursive deletion of workspace. Causing subsequent builds to fail.

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

## Reproduce
Seen frequently on CI

theotherjimmy, 2017-04-21:
Thanks for moving the issue.

theotherjimmy, 2017-04-21:
I'm surprised that windows does not clean up orphaned children by itself. All Linux init process does that.

theotherjimmy, 2017-04-21:
https://wiki.jenkins-ci.org/display/JENKINS/ProcessTreeKiller Might solve your problem

theotherjimmy, 2017-04-21:
https://issues.jenkins-ci.org/browse/JENKINS-26048 Might still be a bug.

mazimkhan, 2017-04-21:
Process tree is like:
Jenkins Slave Service -> mbed-cli with build tools -> compiler

We should not rely on Jenkins features to clean up leaked processes. We should not be leaking them. 
But I will still give this plugin a try :) Thanks!

theotherjimmy, 2017-04-21:
>  We should not be leaking them.

Agreed, when we exit normally. When the user does something like `kill 9` we should not expect to clean up anything.

theotherjimmy, 2017-04-21:
The procces tree is actually:
1. Jenkins slave
   1. mbed-cli
      1. python mbed-os/tools/make.py
         1. compiler
         1. compiler
         1. compiler
         1. compiler

theotherjimmy, 2017-04-21:
The fact that there is a disconnect there might be part of the problem.

mazimkhan, 2017-04-21:
So are compilers spawned in threads? Or ```subprocess.Popen``` objects?
It is hard to imagine that compiler would hung up after compiling a file. Although one thing is possible that it is trying to write on ```stdout``` but we are not reading!

theotherjimmy, 2017-04-21:
> Although one thing is possible that it is trying to write on `stdout` but we are not reading!

We can't if we're dead.

theotherjimmy, 2017-04-21:
> So are compilers spawned in threads? Or `subprocess.Popen` objects?

How would we manage to spawn them in threads? That does not make sense. Anyway, what we do is create a pool of processes, which will call `subprocess.Popen` repeatedly until they drain a common queue.

theotherjimmy, 2017-04-21:
Call chain:
1. https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/__init__.py#L847-L897
1.  https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L192-L203
1. https://github.com/ARMmbed/mbed-os/blob/master/tools/utils.py#L37-L63
1. https://github.com/ARMmbed/mbed-os/blob/master/tools/utils.py#L75-L106

theotherjimmy, 2017-04-25:
I was able to reproduce the problem with [tini](https://github.com/krallin/tini) acting as a sub-reaper on linux.

theotherjimmy, 2017-04-25:
Killing mbed compile with C-c will cause the mbed tools to leak every active compiler process.
e.g.:
```
^C[mbed] User aborted!

[CTRL+c] exit
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7301'
[DEBUG tini (7188)] Reaped child with pid: '7300'
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7274'
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7289'
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7302'
[DEBUG tini (7188)] Reaped child with pid: '7270'
[DEBUG tini (7188)] Reaped child with pid: '7262'
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7287'
[TRACE tini (7188)] No child to reap
[DEBUG tini (7188)] Received SIGCHLD
[DEBUG tini (7188)] Reaped child with pid: '7189'
[INFO  tini (7188)] Main child exited normally (with status '255')
[TRACE tini (7188)] No child to reap
[TRACE tini (7188)] Exiting: child has exited
```

theotherjimmy, 2017-04-25:
@mazimkhan It looks like GCC/ARMCC both fork+exec several child processes, which the build process has no knowlage of.  These are what I mentioned being leaked.

Could you narrow down what causes this failure? I would like to know what caused the processes to be re-parented to the explorer.exe process (which I'm assuming acts like init on linux and reaps children).

theotherjimmy, 2017-05-04:
@mazimkhan Any comments?

mazimkhan, 2017-06-15:
By thread I meant do you Popen from a thread and monitor the process until it finishes. I can see from above link that you are using multiprocessing. I will check the code that you have shared.

theotherjimmy, 2017-06-15:
Yeah, We use mulitprocessing, so not a thread, but another process.

jupe, 2017-06-21:
do we call compiler with `-j` -argument (parallel build)

0xc0170, 2021-01-26:
I'll close this issue as this belongs to old tools.

closed 2021-01-26 by 0xc0170