ARMmbed/mbed-os #4214: I2C block writes and block read bugs

Willem23, 2017-04-21:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32 F103 and probably others

**Expected behavior**
I2C should work properly.

**Actual behavior**
I2C write fails on first Write after reset, Read fails on every other Read.
Ack does correctly report status and returns 1 when the slave is not present.

**Steps to reproduce**
#include "mbed.h"
#define PCF8574_ADDRESS 0x40

int ack;
uint8_t data = 0xAA;
I2C i2c(I2C_SDA,I2C_SCL); 

DigitalOut led1(LED1); 
Serial pc(USBTX, USBRX);

int main() {
  pc.baud(9600);  
  pc.printf("\n\r");
  pc.printf("Start I2C test\n\r");  
      
  led1 = 0;
  wait(0.5);  
  led1 = 1;
  wait(0.5);    
  led1 = 0;
    
#if(0)
  // Set Port Expander
  while(1) {
    wait(1.0);
   
    // Write port
    ack = i2c.write(PCF8574_ADDRESS, (char *) &data, 1);
    data++;
  
    // Check error
    pc.printf("Write Ack = %d\n\r", ack);  
    if(ack != 0) {
      pc.printf("  Write Error\n\r");        
    }      
    led1 = !led1;
  }
#else
  // Get Port Expander
  while(1) {
    wait(1.0);
   
    // Read port
    ack = i2c.read(PCF8574_ADDRESS, (char *) &data, 1);
  
    // Check error
    pc.printf("Read Ack = %d\n\r", ack);  
    if(ack != 0) {
      pc.printf("  Read Error\n\r");
    }    
    led1 = !led1;
  }
#endif  
}

Result for Reads from PCF8574 using mbed rev 133 and higher:
Start I2C test
Read Ack = 0
Read Ack = 1
  Read Error
Read Ack = 0
Read Ack = 1
  Read Error
Read Ack = 0
Read Ack = 1
  Read Error
Read Ack = 0
Read Ack = 1
  Read Error

Result for Writes to PCF8574 using mbed rev 140:
Start I2C test
Write Ack = 1
  Write Error
Write Ack = 0
Write Ack = 0
Write Ack = 0
Write Ack = 0
Write Ack = 0
Write Ack = 0
Write Ack = 0

The I2C lib worked as expected for mbed 132. Revision 133 starts failing for Read. Revision 140 is broken for Write. Have not yet tested at which revision Write starts to break, Read and Write seem to start having issues at different revisions.. 

**Recommend**
Inconsistent behavior should be fixed. Some component Libs now fail because the first Write after reset does not succeed. Some libs fail because they check Ack errors and/or get wrong read values.
See https://developer.mbed.org/questions/77809/Anyone-got-the-STM32F103RB-working-with-/

0xc0170, 2017-04-24:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-04-25:
@Willem23 thanks for reporting the issue and sorry for the inconvenience.
The I2C code was moved into a common code after rev132 and this may be the cause of the issue you're facing. I can now reproduce the problem with the READ. I don't have a PCF8574 but I'm using another slave for test. I can now reproduce the read issue: 1 out of 2 read fails. I can't see the write problem so far. I'll let you know when I better understand.

LMESTM, 2017-04-25:
Note that I made a try with another target (F334R8) and it does not show the problem

LMESTM, 2017-05-15:
@Willem23 Hey - sorry for the delay
I have a branch with a draft fix under review and tests in our HAL team - this may take some time before complete test and feedback. 
https://github.com/LMESTM/mbed/tree/fix_stm32_i2c_pcf
On my test setup, this fixes well the reported issue (but I don't a fake I2C devivce). Would you be able to check on your side and confirm that it fixes your issue ?

bcostm, 2017-05-18:
ST_INTERNAL_REF 33669

LMESTM, 2017-05-22:
Pull request #4365 was sent to fix the issue
@Willem23  If possible, please test and provide feedback.

LMESTM, 2017-06-13:
@Willem23 
#4365 was sent for fixing the issue - if possible, can you please test and provide feedback ?
ST_TO_BE_CLOSED

LMESTM, 2017-07-03:
@0xc0170 @adbridge  I'd suggest to close the issue please
cc @Willem23   - feel free to re-open in case of persistent problem

closed 2017-07-03 by 0xc0170