ARMmbed/mbed-os #4217: Ticker bug on STM32F1

cesarvandevelde, 2017-04-23:
### Description
- Type: Bug
- Related issue: `#816`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32F103

**Toolchain:**
Compiled using PlatformIO 3.3.0
ststm32 @ 1.8.1
framework-mbed @ 3.137.0
tool-stlink @ 1.10200.0

**Expected behavior**
The following code should produce a square wave on pin PA_8:
```C++
#include "mbed.h"

DigitalOut output_pin(PA_8);
Ticker ticker;

void ticker_fn(){
  output_pin = !output_pin;
}

int main(){
  ticker.attach_us(ticker_fn, 500);
}
```

**Actual behavior**
Periodically, the ticker will freeze for 65ms, as shown in the screenshots below.
Immediately following a freeze, all missed interrupts are triggered in a short burst.

[image]
[image]

Some digging reveals similarities to issue #816, though I am unsure how to tackle the problem.

**Steps to reproduce**
The bug occurs on a generic ["blackpill"](http://wiki.stm32duino.com/index.php?title=Black_Pill) STM32F103 board.
The example above was compiled using the following platformio.ini config file:
```ini
[platformio]
env_default = stm32f103c8t6

[env:stm32f103c8t6]
platform = ststm32
framework = mbed
board = nucleo_f103rb
upload_protocol = stlink
```

0xc0170, 2017-04-24:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-04-24:
I thought this issue was solved a long time ago. But I have checked it again on NUCLEO_F103RB (using mbed v140) and I confirm the problem.

LMESTM, 2017-06-05:
@cesarvandevelde @bcostm will be fixed with #4424

LMESTM, 2017-06-09:
ST_TO_BE_CLOSED
#4424 was merged - @cesarvandevelde  the problem is fixed now, feel free to let us know if any further issue

closed 2017-06-09 by 0xc0170