ARMmbed/mbed-os #4218: Cloning takes a lot of time and disk space

nuket, 2017-04-24:
### Description
- Type: Enhancement

Is there any work being done to reduce the amount of time required to run `mbed new` to create a new project and/or reduce the amount of space required by mbed-os? The current checkout takes ~500MB and several minutes to complete. 

Specifically when I'm targeting a single board, it would be nice to have the targets downloaded sparsely when I run `mbed config target`.

bulislaw, 2017-04-25:
I don't think we are working/actively thinking about that at this stage @0xc0170 @theotherjimmy

theotherjimmy, 2017-04-25:
We are not actively working on anything to mitigate this. Why was this marked tools?

bulislaw, 2017-04-26:
Closes fit I could find, but agreed it's not really tools.

cmonr, 2018-01-23:
Closing since https://github.com/ARMmbed/mbed-cli is the more appropriate location for this issue.

Feel free to look over the existing issues there, or open a new issue there.

closed 2018-01-23 by cmonr