ARMmbed/mbed-os #4220: I2C: stuck bus

chrissnow, 2017-04-25:
### Description
- Type: | Question

---------------------------------------------------------------

## Question

How should a stuck I2C bus be cleared in mbed? My current method of a calling NVIC_SystemReset() is a bit extreme!

0xc0170, 2017-04-25:
> How should a stuck I2C bus be cleared in mbed? My current method of a calling NVIC_SystemReset() is a bit extreme!

What errors are you getting ? any specific sensors used? Can you describe the problem more?

HAL defines ``i2c_reset()`` function, however this one is not available through ``I2C`` API. And anyway, it is not yet defined well what reset should do (looking at the docs there).

cc @Sissors @Willem23

chrissnow, 2017-04-25:
I'm going to look in more detail today, all I know at the moment is that reads come back as 0xff and never clears down until a reset is called.

I'm going to handle ACK\NAK better but I still need to know what to do when it happens!

it's a BNO055 and MAX17043, I'm not sure which is the culprit.
It's possible that since the BNO055 reset line gets triggered on host reset that's actually what is fixing the bus rather than the host reset.

In the past on other MCU's I would wiggle SCL by hand until it cleared down.

Sissors, 2017-04-25:
In principle that is for the not too bad stuck cases also what you can do with mbed, create a DigitalOut and toggle SCL until SDA becomes high again, since the typical case of an I2C bus being stuck, is SDA being stuck on low. If this is not the case, then creating a start condition should reset the internal state machine of the slave device. Of course not every slave device does this. Afterwards you can make I2C object again to set the pin in the correct state again (creating a new I2C object is a bit overkill for this, but it is an easy option).

So I could see as option a generic mbed function that switches SCL to GPIO, and toggles it until the bus is cleared again. But if SDA is not stuck low (as in your case, since you are reading 0xFF), then the question is how often should you toggle it?

chrissnow, 2017-04-25:
Having read up on the BNO055 it uses clock stretching, can mbed cope with that? specifically on the xdot which is a STM32L151CC

Having a user accessible i2c reset would be good rather than needing to be implemented by the user.

0xc0170, 2017-04-26:
> Having a user accessible i2c reset would be good rather than needing to be implemented by the user.

Let's look at ``i2c_reset`` , how targets implement it, and how an app can benefit for it? Any proposals appreciated.

chrissnow, 2017-04-26:
How about something based on this.
http://www.analog.com/media/en/technical-documentation/application-notes/54305147357414AN686_0.pdf

> It is the master’s job to recover the bus and restore control to the main program. When the master detects the SDA line stuck in the low state, it merely needs to send some additional clocks and generate a STOP condition. How many clocks will be needed? The number will vary with the number of bits that remain to be sent by the slave. The maximum would be 9. This number is derived from the worst-case scenario, the case where the processor was reset just after sending an ACK to the slave. Now the slave is ready to send 8 data bits and receive 1 ACK (or NAK in the case of a bus recovery).
> 
> The procedure is as follows:
> 1) Master tries to assert a Logic 1 on the SDA line
> 2) Master still sees a Logic 0 and then generates a clock
> pulse on SCL (1-0-1 transition)
> 3) Master examines SDA. If SDA = 0, go to Step 2; if
> SDA = 1, go to Step 4
> 4) Generate a STOP condition
> Note that this process may need to be repeated because the cleared SDA line may have been cleared for the next bit, which was a 1. There may be some concern about the effect this additional clocking and STOPping has on other peripherals. There is no adverse effect; other slaves are not paying attention due to the fact that they have not been addressed. Only the slave that had the interrupted message will respond to the clocks. 
> 
> This procedure is useful in the system code to help restore
> the bus in the event that an SDA = 0 bus fault is
> encountered, regardless of the reason.
> 

The user would call it when they feel it appropriate and it should try and clear the bus down and bring the I2C interface back to a good working state.

Willem23, 2017-05-01:
It is impossible for a healthy I2C master to fix a stuck I2C bus when a rogue slave (or another master !) is pulling SCL low. The only fix is to hard reset the slave or master (at least the I2C engine) that causes the problem.  Some slaves or masters can fix themselves when they have a hardware time-out or watchdog. Some recent NXP devices support the timeout feature in the I2C engine.
This kind of problems could be caused by interrupted I2C messages, by software gone crazy or by power issues on the devices (including I2C level converters 3V3 to 5V etc) or even by defective devices.
Obviously an attempt to fix it by generating extra clocks is better than not even trying. The i2c_reset should return a value showing success or failure.

0xc0170, 2017-05-02:
> Obviously an attempt to fix it by generating extra clocks is better than not even trying. The i2c_reset should return a value showing success or failure.

What would be @Willem23 your proposal for ``i2c_reset`` ? The procedure shared above (https://github.com/ARMmbed/mbed-os/issues/4220#issuecomment-297303467) ?

Willem23, 2017-05-02:
The procedure above is probably best available approach, but should include a check on SCL as well: when SDA is low while master attempts to float it high then try to generate an SCL pulse. Need to check that SCL does indeed go high to ensure that it is not pulled low by the rogue slave.  Return an error in case SCL does not toggle low-high-low, return an error if SDA is still low after max 8 correct SCL cycles. Generate a STOP by floating SDA from low to high while SCL is at high level and return success when both showed correct levels.
Depending on the I2C engine you probably need to temporarily reconfigure the SDA,SCL pins as DigitalInOut (open collector/drain) to make this possible.

saedelman, 2017-07-13:
For NRF5 devices on mBed, the Nordic defined I2C bus timeouts are too short. You have to edit ./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/i2c_api.c and change the TIMEOUT_VALUE from 1000 to 10000 in order to get it working with a uBlox CAM-8MQ as well as the BNO055 because these implement clock streching.

deepikabhavnani, 2017-07-31:
@chrissnow Will it be possible for you to verify I2C recovery fix. I have tested by pulling SDA line low but not with actual I2C device.

https://github.com/deepikabhavnani/mbed-os/commit/32bc86208405270efaa137fbe15188ed8b149fe0

chrissnow, 2017-07-31:
@deepikabhavnani  I'm probably not going to be able to test that in the next few days.
did you look at #3656 ? you should be able to adapt that for any I2C device to test.

I took a look at your code, should we always send at least 8 clocks if SDA is initially low? otherwise you would return if the stuck device clocked a high bit but was still mid transaction?

I think I had this situation as sometimes it would take a few recoveries to clear.

deepikabhavnani, 2017-07-31:
8+1 clock will be the worst case, since we are resetting the I2C line we don't have to explicitly complete transaction. Still if we have issues, can update the code for worst case as well.

I have I2C accelerometer, which is not failing even after resetting multiple times during read. I even tried grounding SDA line, but it recovers after reset (not sure if it has hardware reset).

chrissnow, 2017-07-31:
I think some devices try and recover themselves with a timeout in their I2C handler.

I might be misunderstanding your routine but what if it was mid way through a read of 01010101 when the reset occurred.

would it think the bus was clear after the first high bit was encountered on SDA?

deepikabhavnani, 2018-10-08:
Fixed by #8109

closed 2018-10-08 by deepikabhavnani