ARMmbed/mbed-os #4222: MCUXpresso: warnings in fsl_ headers - enum out of range

0xc0170, 2017-04-25:
### Description
- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Possibly more targets, tested with K64F

**Toolchain:**
ARM

**meed-os sha:**
master as it's today

**Expected behavior**
No warnings

**Actual behavior**

These are the warnings, they are seen multiple times for various compilation units.
```
[Warning] fsl_dspi.h@68,0:  #66-D: enumeration value is out of "int" range
[Warning] fsl_dspi.h@75,0:  #66-D: enumeration value is out of "int" range
[Warning] fsl_dspi.h@82,0:  #66-D: enumeration value is out of "int" range
[Warning] fsl_dspi.h@88,0:  #66-D: enumeration value is out of "int" range
[Warning] fsl_edma.h@145,0:  #66-D: enumeration value is out of "int" range
[Warning] fsl_uart.h@159,0:  #66-D: enumeration value is out of "int" range
```

**Steps to reproduce**
Compile blinky example, should see the warnings

cc @mmahadevan108 - please can you  have a look, or the newer drivers might contain fix ?

mmahadevan108, 2017-04-27:
I have looked at this earlier, this is because some enum values use register bit mask defines with bit 31 set. This is only seen with ARM compiler and not with IAR and GCC_ARM.

0xc0170, 2017-04-28:
> I have looked at this earlier, this is because some enum values use register bit mask defines with bit 31 set. This is only seen with ARM compiler and not with IAR and GCC_ARM.

Thanks, thats correct only with ARM, but it is produced for many files during the compilation. Any proposals for fixing in the upcoming versions?

mmahadevan108, 2017-04-28:
No plans on fixing this. These warnings are currently suppressed in the SDK projects.

deepikabhavnani, 2017-06-12:
@mmahadevan108 - Can we have look at these warnings, we are planning to enable all warnings in mbed-os code base and not suppress them.

deepikabhavnani, 2017-07-27:
@mmahadevan108 - Earlier you commented that these warnings are suppressed in SDK projects.. Wanted to know do u suppress those in Keil packs as well?

mmahadevan108, 2017-07-27:
Do the Keil packs include SDK driver code?

deepikabhavnani, 2017-07-27:
In Keil Pack, i can see NXP -> K60 Series -> MK64xxxx pack -> Device Specific (Older version pack).
I am referring to above pack, if user adds that pack will they have enum warnings?  I am not sure which SDK you were referring to in conversation above.

mmahadevan108, 2017-07-27:
The warnings are from the SDK peripheral drivers files that can be downloaded 
https://mcuxpresso.nxp.com/en/welcome

I am not aware of these drivers being included by Keil. 

Are you seeing similar warning? Please let me know.

0xc0170, 2018-06-19:
@ARMmbed/team-nxp These warnings are still there. See the latest report https://github.com/ARMmbed/mbed-os/issues/7255.

And another duplicate with more information https://github.com/ARMmbed/mbed-os/issues/5082

Is there any change? Won't fix does not help here. Any warning should be eliminated.

maclobdell, 2018-07-27:
Building with K66F also results in these warnings (de-duplicated in this list).  

```
Warning:   Enumeration value is out of "int" range in"   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_clock.h",   Line: 365, Col: 28
--
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_edma.h",   Line: 145, Col: 24
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_dspi.h",   Line: 67, Col: 29
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_dspi.h",   Line: 74, Col: 28
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_dspi.h",   Line: 81, Col: 40
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_dspi.h",   Line: 87, Col: 33
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_lpuart.h",   Line: 142, Col: 29
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_rcm.h",   Line: 110, Col: 23
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_sim.h",   Line: 53, Col: 30
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_sim.h",   Line: 56, Col: 40
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_sdhc.h",   Line: 147, Col: 33
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_sdhc.h",   Line: 234, Col: 32
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_sdhc.h",   Line: 238, Col: 10
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_tsi_v4.h",   Line: 217, Col: 28
Warning: Enumeration value is   out of "int" range in   "extras/mbed-os.lib/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/drivers/fsl_uart.h",   Line: 159, Col: 30
```
@ARMmbed/team-nxp  @mmahadevan108   Please fix as soon as possible.  Thanks!

mmahadevan108, 2018-08-02:
Awaiting a fix from the SDK team. I will update these files as soon as a fix is made available.

maclobdell, 2019-01-14:
@mmahadevan108  - Any update?

0xc0170, 2019-04-12:
It's been a year, @mmahadevan108 a chance for fixing these warnings ?

mmahadevan108, 2019-06-06:
On K66 I see the below warning:
`Compile [ 92.5%]: fsl_flash.c
[Warning] fsl_flash.c@2620,21: unknown attribute 'optimize' ignored [-Wunknown-attributes]
`
Below is the code that is causing the warning, the code is wrapped wit GNUC macro and should not be executed for ARM compiler, can you please explain what is missing.
`
#if (defined(__GNUC__))
/* #pragma GCC push_options */
/* #pragma GCC optimize("O0") */
void __attribute__((optimize("O0"))) flash_cache_clear(flash_config_t *config)
#else
#if (defined(__ICCARM__))
#pragma optimize = none
#endif
#if (defined(__CC_ARM))
#pragma push
#pragma O0
#endif
void flash_cache_clear(flash_config_t *config)
#endif
{
`

0xc0170, 2019-10-24:
Slight delayed msg but `__GNUC__` is also defined for ARMCC in case of Mbed OS. Therefore is better to check also for ARMCC (or first ARMCC and if not defined, and GNUC is, you know its GCC). There are examples in our codebase for this same problem.

0xc0170, 2019-10-24:
I haven't retested latest codebase, @mmahadevan108 these warnings are still valid, aren't they?

mmahadevan108, 2019-10-24:
I will check, can you point me to an example where this is fixed.

0xc0170, 2021-01-26:
I'll close this as won't fix. We will track this independently via new warnings clean-up stories where we provide updated version.

closed 2021-01-26 by 0xc0170