ARMmbed/mbed-os #4227: [LPC15xx] Using UARTs at different baudrates can set conflicting fractional baud rate clocks

devanlai, 2017-04-26:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LPCXpresso1549

**Toolchain:**
All toolchains

**mbed-os sha:**
450701f

**Expected behavior**
Using multiple UARTs at baudrates that are not multiples of each other should generate the correct bit timing for each UART independent of the baudrate of other UARTs.

**Actual behavior**
Creating a second UART can affect the bit timing of existing UARTs, causing the bit timing to be incorrect. The `FRGCTRL` register is shared among all UARTs to set the common UART clock, but `serial_baud` updates `FRGCTRL` only for the most recent baudrate, which can invalidate the timing of any previously initialized UARTs.

**Steps to reproduce**
Sample code with incorrect timing on serial1 (should be 115200):

```
#include "mbed.h"

Serial serial1(D8, NC, 115200);
Serial serial2(D9, NC, 100000);

int main() {
    while(1) {
        serial1.putc('U');
        serial2.putc('U');
    }
}
```

Here's the incorrect output from the above code when creating two serial objects. Note that the bit time is 8.56us, vs the expected nominal bit time of 8.68us.
[image]

Sample code (correct timing, after removing serial2):
```
#include "mbed.h"

Serial serial1(D8, NC, 115200);
int main() {
    while(1) {
        serial1.putc('U');
    }
}
```

Here's the correct output using only a single serial object. Note the bit time is very close to 8.68us, as expected.
[image]

** My comments **
I'm not sure if there's a good way to resolve this issue, since the only register that controls the baud rate that is not shared between all UARTs is `BRG`, which controls the integral clock divider for each UART. This makes it possible to have multiple UARTs that are multiples of the UART clock rate like 9600/19200/57600, but not to mix rates like 115200 and 1000000.

0xc0170, 2017-04-26:
cc @mmahadevan108 @toyowata

toyowata, 2017-04-29:
Hi @devanlai ,

Thank you very much for your detail report.

>This makes it possible to have multiple UARTs that are multiples of the UART clock rate like 9600/19200/57600, but not to mix rates like 115200 and 1000000.

Yes, you are right.  The Serial class HAL implementation of the LPC1549 use U_PCLK for clock source and fractional baud rate generator, so mixing baud rate may have such issue.

[image]

Following table explains current register setting for popular baud rate (and 100000bps).  As you can see, same value of the MULT group (9600/14400/19200/57600) works simultaneously with different USART channel.  This is because each USARTs share single FRG register in SYSCOM.  When baud rate is set by ```Serial::baud()``` API or Serial class constructor with parameter, the FRG value is also calculated/updated to fit proper baud rate.

UARTCLKDIV = 6 (12MHz)

|Baud rate|MULT|BRG|
|:---|:---|:---|
|9600|0x00|0x4D|
|14400|0x00|0x33|
|19200|0x00|0x26|
|38400|0x07|0x12|
|57600|0x00|0x0C|
|100000|0x12|0x06|
|115200|0x15|0x05|
|230400|0x15|0x02|
|460800|0xA0|0x00|

I have investigated to find more suitable value for UARTCLKDEV which is source of fractional baud rate generator.  A value below (UARTCLKDIV=3, 24MHz) could be used for most of popular baud rate include 100000bps.  230400bps and 460800bps are not same grope as others. However, it may be not a problem, since they are not commonly used for serial communication.

UARTCLKDIV = 3 (24MHz)

|Baud rate|MULT|BRG|
|:---|:---|:---|
|9600|0x00|0x9B|
|14400|0x00|0x67|
|19200|0x00|0x4D|
|38400|0x00|0x26|
|57600|0x00|0x19|
|100000|0x00|0x0E|
|115200|0x00|0x0C|
|230400|0x15|0x05|
|460800|0x15|0x02|

You can try to check this using mbed-dev library and change UARTCLKDIV value here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC15XX/serial_api.c#L125

devanlai, 2017-04-29:
Hi @toyowata,
I agree that a UARTCLKDIV of 3 is a better value - I didn't realize that the algorithm was actually calculating a different fractional divider for 9600 baud vs 115200 baud with UARTCLKDIV=6.

There probably is no universal work-around that will minimize the error for all use cases, since there the shared UART clock is a pretty big hardware limitation, but guaranteeing that the standard baudrates up through 115200 won't conflict with each other is definitely a major improvement.

Coincidentally, reducing the divider to 3 also enables baudrates of 921600 and 1000000, which is something else I wanted (when using only one UART), so I'm all for it.

0xc0170, 2017-05-02:
@devanlai Can you send a patch based on the findings here?

cmonr, 2018-01-23:
Keeping this issue open since the source file still shows the clock divisor as six, and changing this issue from a bug to enhancement since this behavior is correct for the part.

deepikabhavnani, 2018-12-28:
@mmahadevan108  - Should UARTCLKDIV be 3 / 6 based on conversation above?

closed 2019-10-31 by ciarmcom

loverdeg-ep, 2019-11-11:
@adbridge Is this not a defect?

adbridge, 2019-11-12:
> 
> 
> Keeping this issue open since the source file still shows the clock divisor as six, and changing this issue from a bug to enhancement since this behavior is correct for the part.
> _[Mirrored to Jira]_

@loverdeg-ep it was marked by Cruz as an enhancement a while back which is why it got automatically closed by the bot. If you feel it is in fact a defect (despite Cruz's comment) then we can  reopen it and tag it as such....

peterkwek, 2020-01-22:
I had also the problem for the LPC54114 device, that I need multiple baudrates like 19200, 115200, 230400 and 460800, not sure if my solution works also on other devices.

A couple of these can be used, all together is hard to use with the fractional rate generator.
So I looked which other CLK sources can be used for using directly by the flexcom interface.
For example the fixed FRO_12, also for this frequency it's hard and not possible to get all these baudrates. Same for FRO_HF not, in my case 48Mhz.
mclk_in is not connected for me, so also not an option. 

THe PLL_CLK the only remaining, can be configured to any frequency and can be used as clock source for the flex_COMM to generate baudrates.
Also very easy to do it.
Make a list of baurdates you like, choose a fixed value of OSRVAL, and change BRGVAL, untill all baurdates have the same CLK value. 
User manual gives formula for baudrate:   baud rate = [FCLK / (OSRVAL+1)] / (BRGVAL + 1)
Re-written like: FCLK = "baud rate" x ( BRGVAL + 1) x (OSRVAL + 1).
Fill in values and find your PLL_CLK frequency.
Setting up your PLL_CLK first.
Values can be used fixed, or make some functions to calculate BRGVAL from baudrate input and fixed PLL_CLK and fixed OSRVAL.