ARMmbed/mbed-os #4237: "new" intel hex 03 record block flashing by Nordic cmsis-dap interface

nvlsianpu, 2017-04-27:
## Description
I encountered  new problem Nordic cmsis dap is unable to load hex file to the target (using MSD). I investigated: In *.hex appeared new record `:04000003200026555E` - which is source of this issue.

@0xc0170 could you investigate or ask someone to looka at this?

mbed-os master 12ca90e71f4fdce08240e0c706310afa4f928e97

## step to reproduce:
build any of nordic target eg
`mbed compile -m NRF52_DK -t GCC_ARM`

0xc0170, 2017-04-27:
Unable to load? any error shown or? How does this behave using IDE when loading .elf file - works?

cc @c1728p9

c1728p9, 2017-04-27:
Hi @nvlsianpu, what version of DAPLink are you using? Can you attach details.txt from DAPLink to this issue? Also, could you attach the hex file you are seeing this failure with?

nvlsianpu, 2017-04-28:
Regular bluetooth application (mbed compile -m NRF51_DK -t GCC_ARM -c0 https://github.com/nvlsianpu/miscellaneous/blob/mbed-os-hex/BLE_central.hex This one has 03 record just at the beginning.

Test (mbed test -m NRF51_DK -t ARM -n "mbed-os-tests-mbed_drivers-c_strings" -c) https://github.com/nvlsianpu/miscellaneous/blob/mbed-os-hex/c_strings.hex This one has 05 record just at the beginning.

used CMSIS-DAP tag: https://github.com/NordicSemiconductor/CMSIS-DAP/tree/V0221

I mentioned MSD interface so I just daragged and drop a file to mbed storage driver - therefore I don't have a text log.

nvlsianpu, 2017-04-28:
@c1728p9, @theotherjimmy:
This [change] (https://github.com/ARMmbed/mbed-os/commit/74998d649abda00db92716f5a3a53d25cad95737) cause this issue - probably as side effect.

c1728p9, 2017-04-28:
Hi @nvlsianpu, you are using an old version of interface firmware which doesn't support this record type:
https://github.com/NordicSemiconductor/CMSIS-DAP/blob/V0221/interface/Common/src/intelhex.c#L313

If you update to the latest interface firmware, DAPLink version v0243, hex file programming should work since the hex file parsing handles this record type. I was able to load both versions you had attached with no problems with it. For reference this release can be found as a [release 0243 in the DAPLink repo](https://github.com/mbedmicro/DAPLink/releases/tag/0243). You'll want to update your interface with 0243_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin.

nvlsianpu, 2017-05-04:
This interface firmware works well with nrf52_dk. but Unfortunately it doesn't work well with nRF52840_DK. Is it possible to fix this issue on mbed-os side? I ask because fixing this on Nordic side will take much longer amount of time because we will need to switch from CMSIS-DAP to DAPLink sources.

cc @David-Garcia-Polo

nvlsianpu, 2017-05-05:
I have resolution for this issue: #4276

theotherjimmy, 2017-05-08:
#4276 Merged. Closing.

closed 2017-05-08 by theotherjimmy