ARMmbed/mbed-os #4244: ublox_evk_odin_w2 WiFi broken on master

bulislaw, 2017-04-28:
### Description
- Type: Bug 
- Related issue: https://github.com/ARMmbed/mbed-os/pull/4238
- Priority: Major

## Bug

**Target**
ublox_evk_odin_w2

**Toolchain:**
GCC_ARM

**Toolchain version:**
6q1

**mbed OS version**
master `12ca90e71f4f`

** Description **

The https://github.com/ARMmbed/mbed-os-example-wifi fails to connect to WiFi network using Odin board, scan works, but the board hangs on connecting. I've tried multiple network both open and close.

@andreaslarssonublox @andreaspeterssonublox

0xc0170, 2017-04-28:
> 6q1

One thing to be certain, I believe the ublox built the libs with gcc arm 4.9.x version.

andreaslarssonublox, 2017-05-03:
@bulislaw I just tested it here on today's master and think I had the same problem. I looked at the code and saw that sleep mode is enabled for our target. I  don't think it's compatible with the way the driver works right now so can you try to remove SLEEP from targets.json and test again to see if it's the same problem I have?

bulislaw, 2017-05-09:
Fixed in #4270

closed 2017-05-09 by bulislaw