ARMmbed/mbed-os #4246: Can't use TCPSockets over 6LoWPAN on others targets than K64F

pmancele, 2017-04-28:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
STM32L476

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.0.0

**Hardware**
Firefly 6LoWPAN shield + STM32L476ZG
 
**Expected behavior**
We are using the last version of mbed-os and tried to run a simple easy-connect example using 6LoWPAN. It worked well on the K64F, we can create a NetworkInterface, open a TCPSocket, connect to website ("linkedin.com) and send a simple HTTP GET request.

But we did not achieve to make this work on any other MCU. Our main target is a STM32L476ZG. 
With the same program, it connects sucessfully to the 6LoWPAN network, we can create a socket but we cant  connect to any website as the DNS requests are not working.

Both the K64F and the STM32L476 have the same main:
 ```c
TCPSocket socket;
NetworkInterface* LoWPAN;
int main()
{
	LoWPAN = easy_connect(true);
	ret = socket.open(LoWPAN);
	if (ret != NSAPI_ERROR_OK) {
		printf("Error opening socket : %d", ret);
	}

	//ret = socket.connect("2620:109:c002::6cae:a0a",80);
	ret = socket.connect("linkedin.com",80);
	printf("nsapi_error : %d \n", ret);

	if (ret == 0) {
		printf("It's OK ! \n");
	}
	else {
		printf("Failed ! \n");
	}
}
```
**K64F output:**

> [EasyConnect] Using Mesh
[EasyConnect] Connecting to Mesh..
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 2001:470:1f13:af3:fec2:3d00:4:583
[EasyConnect] MAC address fc:c2:3d:00:00:04:05:83
nsapi_error : 0 
It's OK !

**STM32 Output:**

> [EasyConnect] Using Mesh
[EasyConnect] Connecting to Mesh..
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 2001:470:1f13:af3:fec2:3d00:4:583
[EasyConnect] MAC address fc:c2:3d:00:00:04:05:83
nsapi_error : -3009 
Failed

The **-3009** error is a DNS error. We debugged the program and the DNS packet seems to be built correctly but never sent by the firefly 6LoWPAN shield. 
```err = socket.sendto(SocketAddress(dns_servers[i], 53), packet, DNS_BUFFER_SIZE);``` indicate us that the packet has been transmitted, but it seems it hasnt  as we cant see it with whireshark sniffing our IPv6 network.

**If we use *socket.connect()* with an IPV6 address instead of a website address it is working both on K64F and STM32L476..**

Finally we used mbed-trace both on the K64F and on the STM32L476. On the K64F there is no error reported by mbed trace during all the process, but on the STM32 we have this **ERROR** occuring several times:
```
[ERR ][mMCp]: MAC tx fail

[DBG ][rplc]: Triggering parent selection due to ETX change on fc:c2:3d:00:00:04:df:8b

[ERR ][6lAd]: MCPS Data fail by status 233
```

How to fix it ? Have you any idea ?

We tried to use the same 6LoWPAN shied on NUCLEO_F401RE and NUCLEO_L152RE to try DNS resolution but we cant get it working. EasyConnect return -3012 error which is *NSAPI_ERROR_DEVICE_ERROR*.

0xc0170, 2017-04-28:
cc @kjbracey-arm @SeppoTakalo

kjbracey-arm, 2017-05-02:
The errors suggest the radio is having problems - apparently its transmissions are not being acknowledged, and from your Wireshark trace, it appears it's the transmit getting lost, not the acks.

 It's possible that it's due to a lack of memory, given that it managed to get onto the network, and only failed DNS later. But the RAM in that device should be more than adequate for 6LoWPAN. Maybe check the `mbed-mesh-api.heap-size` config setting?

On some boards there can be pin conflicts between on-board peripherals and the Arduino riser - could be that too.

I don't believe we've tested that particular board+radio combo yet, but we do have a STM32L476 board to hand - we'll can try it.

kjbracey-arm, 2017-05-02:
Do the two Nucleo boards get anywhere at all? Is that an immediate connection failure?

@JanneKiiskila, is there any K64F-specific magic for 6LoWPAN in EasyConnect? Any reason why it shouldn't work with the Atmel radio module on a different board?

JanneKiiskila, 2017-05-02:
jankii01@ubuntu:~/mbed$ cd mbed-cloud-client-example
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ cd easy-connect/
jankii01@ubuntu:~/mbed/mbed-cloud-client-example/easy-connect$ git grep -i K64F
jankii01@ubuntu:~/mbed/mbed-cloud-client-example/easy-connect$ 

empty

There isn't a single reference to K64F in the easy-connect repo, so there's no magic there.
https://github.com/ARMmbed/easy-connect

pmancele, 2017-05-02:
@kjbracey-arm Hello and thank you for your answers.

Yes we thought it was a memory issue so we tried to modify the heap size with `mbed-mesh-api.heap-size` without success. We tried `10000` and `30000` of heap-size. (Btw its working with 30k on K64F but not with 10k)
We will check again for a pin conflict or bad pins configurations but we already tried it..

If you have a STM32L476 we are definetly interested for you to test the combo !

@JanneKiiskila Yes easy-connect seems working fine on our STM32L476ZG, as we can connect and get an IPv6, but the issue occurs while using the TCPSockets and DNS resolutions.

Therefore, on NUCLEO boards we cant connect with easy-connect, and it returns the same error that is returned when the shield is not plugged on the boards. So on nucleo it seems to be a pin issue, but all the arduino pins are well declared on the PinNames.h file and seems to be compatibles with the required functionnalities.

0xc0170, 2017-05-02:
cc @bcostm @adustm You might be interested in this one.

kjbracey-arm, 2017-05-02:
The pins may be well declared in PinNames.h, but there can easily be a clash with the debug UART or something else. Sometimes a link might need changing on the board to make the declared pins reach the Arduino header.

I know we've had a problem with at least one Nucleo board not being useable with that Atmel radio. I forget what the precise pin clash was, but it wasn't possible to usefully work around.

(Minimum useable mesh heap size for a 6LoWPAN router should be something in the range 15K-20K).

adustm, 2017-05-02:
Hello, just for my understanding : @pmancele I remember you are not working on the NUCLEO_L476RG but on another platform based on STM32L476ZG. Correct ?

pmancele, 2017-05-02:
Yes we designed our own plateform based on a STM32L476ZG that we ported to mbed-os.
We also get recently the new STM32L496ZG but we have the exactly same issue.

We also have official mbed boards like K64F, NUCLEO F401RE and L152RE but we successfully tried 6LoWPAN on K64F only (with the atmel firefly shield).

adustm, 2017-05-02:
Hello @jeromecoutant 
Could you try the main code mentionned above with NUCLEO_L476RG + firefly 6LowPan shield ?
cheers
Armelle

pmancele, 2017-05-03:
@kjbracey-arm Here is what trace prints while trying to solve DNS:
```
[DBG ][nsif]: socket_sendto(socket=0x2000a814) sock_id=2, ret=512
[DBG ][nsif]: socket_recv(socket=0x2000a814) sock_id=2, ret=-3001
[ERR ][mMCp]: MAC tx fail
[ERR ][6lAd]: MCPS Data fail by status 233
```
This is when a UDPSocket is created to send and receive the DNS packet. (But as i explained we dont see this DNS packet with wireshark)
The **-3001** error of the socket.receive is NSAPI_WOULD_BLOCK

kjbracey-arm, 2017-05-03:
It may be that the radio is unusually unreliable.

I note that the nsapi_dns.cpp implementation makes no attempt to retry - it does only one transmission, and gives up if it doesn't get a response in 5 seconds. So it would be more vulnerable to unreliability than the network bringup or the TCP connection itself.

kjbracey-arm, 2017-05-03:
(That was 1 UDP/IP transmission - the IEEE 802.15.4 MAC layer should perform about 5 transmissions itself though, looking for an ack)

pmancele, 2017-05-03:
@jeromecoutant @adustm I just got a NUCLEO_L476RG and tried on it. First it was impossible to connect to the shield due to a pin conflict with the LED. (PA_5).
After solving the issue i got the same behavior as on our STM32L476ZG. DNS resolution failed and the trace is still printing a lots of errors:
```
[ERR ][mMCp]: MAC tx fail
[DBG ][rplc]: Triggering parent selection due to ETX change on fc:c2:3d:00:00:04:df:8b
[ERR ][6lAd]: MCPS Data fail by status 233
```

kjbracey-arm, 2017-05-03:
Do the same MAC tx errors occur during the network bringup? Or do they start later?

pmancele, 2017-05-03:
@kjbracey-arm yes it happened before the network bringup too, like 4/5 times.
But only on STM Boards. I will try to analyse the SPI signal. Maybe there is a SPI driver issue on STM32L4 ?

kjbracey-arm, 2017-05-03:
Certainly possible. I don't know how much the SPI bus is ever exercised in our platform tests, if at all.

Not sure the symptoms match that though - if the radio didn't recognise a transmit command due to SPI corruption, the lack of a "TX finished" IRQ would likely seriously upset the driver - possible that it's quietly timing out though?

If the SPI bus just corrupted the data payload, the radio would end up computing a valid MAC checksum, so you'd get an ack and see the UDP packet with incorrect UDP checksum heading onto the Ethernet (the UDP checksum would not be checked until the destination).

Do you have an IEEE 802.15.4 sniffer?

pmancele, 2017-05-03:
Thank you for the explanations. 
No we dont have sniffers..

SeppoTakalo, 2017-05-03:
Probably unrelated: How far away are the devices from each other?
TX errors might be that they are in edge of the range that they can operate. These devices use less power than Wifi and simpler error corrections. With poor antenna, the range can be as low as 10 meters.

pmancele, 2017-05-04:
@SeppoTakalo They are very close (1 meter) so i dont think this is related

@kjbracey-arm We have an IEEE 802.15.4 sniffer now. While using the K64F we can see the UDP packets on port 53, corresponding to the DNS request.
While using the L476 we dont have this packets. However there is a lots of packed sniffed (including UDP packets but not on port 53) but i'm afraid that my network skills are not enough to understand what is happening...

@adustm I used a logic analyzer to monitor the SPI connection and I discovered and issue with SPI on stm32 boards. The spi connection is well established but the frequency required by the mbed-os function spi.frequency() is never reached..
By default it is 1MHz. On STM32L476ZG and nucleo F401RE we only reach 650 kHz !
While requesting 7.5 MHz (which is the frequency asked by the ATMEL 6LoWPAN driver) it reach about 4.9 MHZ..
Should I create an issue about this ?

kjbracey-arm, 2017-05-04:
Looking at the STM implementation, it can only achieve approximate frequencies by dividing by powers of 2. So 4.9MHz may be the closest it can manage to 7.5MHz. As long as it's not dropping below 3.75MHz, it's probably working as intended.

This would reduce performance a bit, but I wouldn't expect a major problem.

This problem seems remarkably specific, if you are consistently seeing no DNS requests via the sniffer, but you are managing all that other traffic to get the network up.

It reminds me of something we saw a while ago where one particular CoAP query consistently failed to respond on a platform, but everything else seemed to work fine. 

The reason for that turned out to be that there was a radio driver/hardware bug which it failed to send packets with one specific length. So almost all traffic worked, but that particular CoAP request consistently failed, due to its length.

Can you try changing the length of the DNS request and/or requesting different addresses? Maybe try manually sending a few similar but slightly-different UDP packets? What's special about the DNS request?

kjbracey-arm, 2017-05-04:
Is it just excessive size? Is this the largest thing that the node has sent? The first 6LoWPAN fragmented packet, say? Try simple UDP sends of various sizes.

pmancele, 2017-05-04:
@kjbracey-arm Thank you for your answer ! I will try what you suggest.

pmancele, 2017-05-04:
@kjbracey-arm I'm sorry but during the DNS request, we actually got 6LoWPAN packets emitted by the STM32L476 that we sniffed.
Attached, you can find two 802.15.4 capture files. One is corresponding to the K64F run, and the other is the STM32L476.
we captured all the process : connection with easy-connect --> TCPSocket creation --> TCPSocket connection to "linkedin.com" which should resolve DNS.

You can open it with wireshark and compare the 6LoWPAN packets corresponding to the DNS request.
Packets 74 to 88 on K64F and packets 102 to 114 on STM32L476

On K64F the 5 6LoWPAN packets from 81 to 85 are **differents** and correspond to the UDP DNS request.
On STM32L476 the 6LoWPAN packets emitted are all the **same**...

[IEEE_802.15.4_capture.zip](https://github.com/ARMmbed/mbed-os/files/976220/IEEE_802.15.4_capture.zip)
 
I can take screenshot for you if you dont have access to wireshark now but want to take a closer look on it ?

pmancele, 2017-05-04:
We can see the same behavior on the SPI link: The K64F is writing different packets to the ATMEL buffer, but the STM32L476 is always sending the same packet to the ATMEL buffer.

The packets send by the STM32L476 are similar to the first one sent by the K64F, and contains the string "linkedin.com"

SeppoTakalo, 2017-05-05:
In K64F capture.
Packets from 81 - 81 are fragmented send. So the DNS query in packet 86 is actually requiring 6 packets to send through. Wireshark shows the last fragment as a complete packet.

Ack packets seems to be missing for most of the fragments, maybe sniffer did not catch them.

Similar sequence is visible in STM32L476 capture in packets 110 - 114. But instead of separate fragments, it seems to be sending the same packet again and again.
It match the behaviour you desribed earlier, same packet is visible in the SPI wire.

If I had to guess, there might be something wrong with the SPI driver as it keeps repeating the previous transfer, if it is timed close enough.

SeppoTakalo, 2017-05-05:
Actually, it can also be retransmissions as the ACK packets are not shown.

kjbracey-arm, 2017-05-05:
Looking more closely at the traces.

The K64F is showing some comms problems. Packets 74-79 are a DNS attempt, and packet 75 is apparently unacknowledge, so it is transmitted 5 times. 5 seconds later packets 81-86 go through without error - the node apparently saw acks, but the sniffer didn't.

Other retransmitted packets include 40-47. There we can see the same packet transmitted 5 times, but we can actually see the acks in the sniffer.

This suggests some significant radio interference problems. It could be transmission power of the acks - sometimes the node misses them, sometimes the sniffer misses them. The Atmel radio does have a cunning power saving feature which makes it transmit acks quietly if it's acking a packet it received loudly. We disabled this in the driver a while ago as it performed badly. It's possible the gateway binary still does that.

The STM trace doesn't to me suggest SPI problems - you just see the first fragment transmitted 5 times (including being resent over the SPI), then it gives up, due to not hearing an ack.

I note that the DNS is sending pointlessly big packets. It's doing sendto(DNS_BUFFER_SIZE) in nsapi_dns.cpp, rather than the actual size of the query.  Fixing that might help a bit.

kjbracey-arm, 2017-05-05:
I also see some checksum failures (wpan.fcs.bad), eg K64F packet 93. Although that was acked in 94 indicating that the node received it correctly, even though the sniffer didn't.

kjbracey-arm, 2017-05-05:
Generally the radio looks bad enough that I'm wondering if you're just suffering a significant channel collision with your Wi-Fi.

pmancele, 2017-05-05:
@kjbracey-arm Thank you !! 
We tried the modification you suggested #4274 and it solved the DNS issue on our STM32L476ZG !!
We will use the sniffer and the logic analyzer to see if all the problems are solved, but for now this modification solved the DNS resolution issue !
I'll keep you in touch.

adustm, 2017-05-05:
Hello @pmancele 
I've just studied your SPI values.
I guess you are using the SPI pin of the arduino connector, so SPI1 on NUCLEO_L476RG. This SPI is connected to the APB2 bus. 
If you reach 650MHz while requesting 1MHz, I guess you are using the HSI clock at 80MHz and not an external clock (different with the NUCLEO board).
On the Nucleo we are using 
In your case, APB clock is at 80MHz. The SPI clock can then be set at 

The available values of SPI clock are between APBclk/2 and APBclk/256
```
clock choice                  | SPIpresc=0 | SPIpresc=1 | SPIpresc=2 | SPIpresc=3 | SPIpresc=4 | SPIpresc=5 | SPIpresc=6 | SPIpresc=7
------------------------------|------------|------------|------------|------------|------------|------------|------------|--------------
HSI clk with APB2clk at 80MHz |   40MHz    |   20 MHz   |   10MHz    |   5 MHz    |   2.5MHz   |   1.25MHz  |   625kHz   |   312 kHz
HSE clk with APB2clk at 48MHz |   24MHz    |   12 MHz   |   6MHz     |   3MHz     |   1.5MHz   |   750kHz   |   375 kHz  |   187kHz
```
Our sw will check the APBclock value, and provide the closest frequency below the requested one.

Maybe you could change you system clock so that you have a better SPI clock. But take care that if you are using USB for instance, USB will need a 48MHz clock.

Hope this helps.
Armelle

kjbracey-arm, 2017-05-05:
Clearly that doesn't solve the real problem - the network is apparently still going to apparently behave quite poorly when you send any significant traffic. But at least it sidesteps it at that early DNS stage. And maybe if the problem is asymmetric you might continue to side-step it with asymmetric traffic, with the STM never transmitting much.

You can continue to test with some manual UDP sending.

pmancele, 2017-05-05:
@adustm Yes that is exactly what we were trying to do, thank you for the precision.
Yes we had a system clock at 80Mhz. We used the PLL clocked on MSI which is set at 48Mhz (as it is done for the STM32L476RG on mbed-os)
We tried to modify the `SetSysClock_PLL_MSI()` to get 60MHz system clock instead of 80 (Indeed 60/8 = 7.5Mhz for our SPI bus).
So we modify the function as this:
```
RCC_OscInitStruct.PLL.PLLM            = 4;
RCC_OscInitStruct.PLL.PLLN            = 10;
RCC_OscInitStruct.PLL.PLLP            = 7;
RCC_OscInitStruct.PLL.PLLQ            = 2;
RCC_OscInitStruct.PLL.PLLR            = 2;
```
We did get a 60Mhz clock but our SPI3, (which is on APB2 bus), is not starting at all with this new clock configuration...(uart is still working though)

@kjbracey-arm Indeed. Our current objectives is to established a secured connection (mbed-tls) with AWS-IOT using 6LoWPAN, and to communicate using MQTT. We already made this working on the K64F.

kjbracey-arm, 2017-05-05:
I suspect if you can't send a 512-byte DNS packet you're probably not going to be able achieve a complete TLS/DTLS handshake. The handshake is pretty intense.

kjbracey-arm, 2017-05-05:
Diagnostic tip - the rank indicated in RPL DODAG Information Object packets tells how badly the nodes think they're doing at radio comms overal, as an ETX value - see RFC 6550 and RFC 6719.

Not only are both traces showing bad values, they're sometimes showing 0xFFFF, which indicates a worse-than-minimum-acceptable link.

pmancele, 2017-05-10:
When monitoring the SPI link between the NUCLEO_L476RG and the ATMEL 6LoWPAN shield, we saw that when we have to send something, the STM32L476 is always writing 5 times the same buffer while the K64F is only doing it once.
We can also illustrate this issue by pinging the STM32L476:
**5 pong are received for each ping !**
[image]

When reducing the SPI speed on the K64F we get the same behavior ! That is to say the K64F can connect to a network, open sockets but every buffer to transmit is send 5 times on the SPI link.

That's why we thought it was a SPI speed issue on the NUCLEO_L476RG. Previously the SPI speed was 5Mhz. We reduced the systemcoreclock to 60MHz in order to reach 7.5Mhz on the SPI. But it doesn't fix the problem..

We tried the mbed-os-example-client on K64F and NUCLEO_L476RG

**Output of K64F:**
* systemcoreclock is **120MHz**
* SPI speed is 7.5MHz
* I2C frequency is 100kHz

> Starting mbed Client example in IPv6 mode
[EasyConnect] Using Mesh
[EasyConnect] Connecting to Mesh..
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 2a01:e35:2e2c:4440:fec2:3d00:3:19dd
[EasyConnect] MAC address fc:c2:3d:00:00:03:19:dd

> SOCKET_MODE : UDP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684

> Registered object successfully!
handle_button_click, new value of counter is 1
handle_button_click, new value of counter is 2

Output of the **NUCLEO_L476RG:**
* systemcoreclock is **60MHz**
* SPI speed is 7.5MHz
* I2C frequency is 100kHz

> Starting mbed Client example in IPv6 mode
[EasyConnect] Using Mesh
[EasyConnect] Connecting to Mesh..

> Starting mbed Client example in IPv6 mode
[EasyConnect] Using Mesh
[EasyConnect] Connecting to Mesh..
[EasyConnect] Connected to Network successfully
[EasyConnect] IP address 2a01:e35:2e2c:4440:fec2:3d00:3:9ccd
[EasyConnect] MAC address fc:c2:3d:00:00:03:9c:cd

> SOCKET_MODE : UDP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684
simulate button_click, device not registered
simulate button_click, device not registered

Currently we are looking for a timeout issue in the ATMEL rf driver that could interrupt the ACK process and restart the transmission process.

kjbracey-arm, 2017-05-11:
There is one speed-critical part of the driver - the reception is run in "manual" mode which means the driver has to process ack reception itself. The chip has to be turned around from TX to RX mode via interrupt after receiving the "TX done" interrupt.

A slow SPI bus and/or slow interrupt response could interfere with that, leading to missing the ack, causing those symptoms.

The driver has to get the chip into RX mode within 192us of the TX done interrupt. A bit of scope/logic analysis should show whether it's missing that. The code path is  rf_if_interrupt_handler (IRQ) -> rf_if_irq_task (RealTime thread) -> rf_if_irq_task_process_irq -> rf_handle_tx_end -> rf_receive -> rf_if_change_trx_state(RX_AACK_ON).

However, the sniffer traces shown above show that the sniffer is also struggling to detect acks - sometimes the sniffer gets the ack, and the node does, and sometimes the node does but the sniffer doesn't. On both the K64F and STM traces. That may suggest there's another problem. Or it could be a poor sniffer - we've certainly seen some quite unreliable sniffers.

kjbracey-arm, 2017-05-11:
Last time I looked at this (which was when moving the work from being directly in IRQ to a top-priority thread), the K64F was consistently managing the turnaround comfortably under 100us. The speed of the SPI bus probably dominates.

If getting the SPI up to 7.5MHz in the STM doesn't solve it, maybe there's some other problem with the SPI driver.

Another possibility - there are nanosecond software timing loops in the driver too. If `SystemCoreClock` were somehow set higher than the real CPU speed, these delays might be coming out too large.

pmancele, 2017-05-11:
@kjbracey-arm Indeed this seems to be the source of the problem !

On the K64F it elapsed 144µs between the TX_DONE interrupt and the state changement to RX_AACK_ON.
Then we have a new TRX_END interrupt, telling to the MCU that something has been received (the ACK)

On the STM32L476, it takes **344µs** with systemcoreclock at 60Mhz and SPI at 7.5 MHz.
With systemcoreclock at 80MHz and SPI at 5MHz it takes **278µs.**
Because of this we are missing the ACK, and we dont get the TRX_END interrupt that should occured when the ACK is received.

I join you 3 screenshot of the result given by the logic analyser. The windows is focused on the turnaround part, between marquers A1 and A2, duration shown in the top right corner. Channel 4 is the Interrupt pin:

* 1st screenshot is the K64F turnaround (144µs)
[image]
* 2nd screenshot is the STM32L476 turnaround at 80MHz with SPI at 5 MHz (278µs)
[image]
* 3rd screenshot is the STM32L476 turnaround at 60 MHz with SPI at 7.5 MHz (344µs)
[image]

Should i open an issue on the ATMEL RF driver repository to indicate that it takes too much time to realize the turnaround on other boards than the K64F (or at least on ST boards with max systemcoreclock of 80MHz) ?

kjbracey-arm, 2017-05-11:
Good investigation - well done!

Sure, you can open an issue on the Atmel driver repo.

Could you show a more zoomed in trace on the initial SPI transactions - same scale for all 3? I'd like to compare exactly where the extra time is going.

Why should the 80MHz/5 MHz one be faster? The slower SPI should make it slower. Is the SystemCoreClock variable correct? Maybe check delay_ns() calibration with the scope? Or is there just so much code in the SPI driver that CPU speed dominates?

Assuming the time is mainly going on the SPI, I have a feeling we may need to extend the mbed OS SPI API to fix this - it lacks a synchronous block transfer call, and the STM HAL is clearly not designed for byte-at-a-time use. If we could reduce the number of HAL calls by a factor of 4 or so, doing multi-byte transfers, maybe that would cover it.

The existing asynchronous block transfer API is overkill in terms of driver complexity for these small commands.

kjbracey-arm, 2017-05-11:
I am also slightly surprised to see the K64F taking so long. I'm pretty certain it was doing better than that. @c1728p9 do you by any chance recall the numbers? I recall it was your scope hooked up to it that time.

pmancele, 2017-05-11:
Here is a zoom on the first initial transactions, with the same scale (50µs long). It seems that the issue is not the SPI speed but the time between each SPI transaction which is too long on STM boards, and seems to depend on the systemcoreclock.

* K64F at 120 MHZ, with SPI at 7.5 MHz:
[image]
* STM32L476 at 60MHz, with SPI at 7.5MHz:
[image]
* STM32L476 at 80MHz, with SPI at 5MHz:
[image]

kjbracey-arm, 2017-05-12:
Okay, that set-up time is huge. If we were to offer a multi-byte transfer API, for those few accesses, it would only help a bit. The register accesses are only 2- or 3-byte bursts. It looks like it would still take too long.

There is also the curious fact that the 60MHz version is not proportionately slower. Surely it should take 4/3 as long as the 80MHz between transfers, not virtually twice as long? Does that indicate some problem?

My approach to tackle this would be three-pronged:

  * Add multi-byte synchronous master transfer API to mbed HAL
  * Update Atmel driver to use this API
  * Request ST to do whatever they can to reduce their SPI transfer setup time.

There may also be some scope to reduce number of transfers in the driver's critical path (eg use the option that gives automatic status output during the register address byte) but without some improvement from ST, I don't see we'll ever get to 192µs.

The alternative "last resort" would be a driver change that allows it to use automatic mode, but that's non-trivial and we don't want to lose the manual mode capability, so it would increase driver complexity not reduce it. (The manual mode permits more tuning of transmission timing, and we've found the 802.15.4 specced behaviour used by the hardware doesn't work well in some environments that we target with the Atmel chip).

LMESTM, 2017-05-15:
@pmancele @kjbracey-arm I've pushed a branch for test which limits the registers access in STM32 SPI driver. This is not fully tested yet - but any feedback or testing from your side would be more than welcome.
https://github.com/LMESTM/mbed/tree/STM32_SPI_LL_USAGE
Note that this would take more time to actually deploy on all STM32 families and make a complete non-regression testing.

LMESTM, 2017-05-15:
One more note: testing the patch requires to manually define of STM32_SPI_LL_USAGE
details here https://github.com/LMESTM/mbed/commit/c02d424b662ffd0eac61856f0794dbc00c7120e7

pmancele, 2017-05-16:
@LMESTM I've tested the patch and there is a big improvement indeed.
With the STM32L476 at 80MHz, it manage to realize the turnaround within 125µs (VS 278µs previously) when i compile with the RELEASE profile. But if i compile with the DEBUG profile it takes 245 µs which is still too much.
Moreover, for lowpower application, the clock may be under 80MHz so we still need to improve the communication by grouping the transfer as much as possible!

Nevertheless, thank you for this improvement!

* STM32L476 at 80 MHz, RELEASE profile (125µs):
[image]
*STM32L476 at 80MHz, DEBUG profile (247µs):
[image]

kjbracey-arm, 2017-07-06:
I believe the core problem here is resolved by ARMmbed/mbed-os#4375. Can you confirm?

pmancele, 2017-07-06:
It is resolved yes, but only if you build with the release profile as explained pleviously. 
With some driver improvements (which are on going as you explained here https://github.com/ARMmbed/atmel-rf-driver/issues/53#issuecomment-313344029) we may be able to fully resolve the problem.

kjbracey-arm, 2018-06-15:
I believe this was all resolved. Reopen if not.

closed 2018-06-15 by kjbracey-arm