ARMmbed/mbed-os #4250: USB Device support broken for K64F

screamerbg, 2017-04-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Minor (as USB is unsupported, though someone might want to investigate and fix)

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

**Toolchain version:**
5.06

**mbed-cli version:** 
1.1.1

**meed-os sha:**
mbed-os-5.4.4 release

**DAPLink version:**

**Expected behavior**
Trying USBMouse HelloWorld example from the mbed Developer Site, where mbed library is replaced by mbed-os. For STM32 and Nuvoton boards the steps to reproduce below work just fine both with mbed Classic library (rev 140) and mbed-os (5.4.4).

**Actual behavior**
Example program compiles fine but getting USB Device not recognized on Windows 7. Having 2x K64F therefore is highly unlikely tobe HW bug.

**Steps to reproduce**
1. Import USBMouse_HelloWorld https://developer.mbed.org/users/samux/code/USBMouse_HelloWorld/ (code dates back to 2013!)
2. Remove USBDevice library
3. Add/clone my fork at https://developer.mbed.org/users/screamer/code/USBDevice/ and switch to branch stm32-support, which is identical to mbed-os/features/unsupported/USBDevice
4. Update mbed library or replace with mbed-os
5. Compile in the mbed Online IDE or `mbed compile -t ARM -m NUCLEO_F429ZI`
6. Flash the board and then change USB connection from STLink to USB Device port

0xc0170, 2017-04-29:
cc @mmahadevan108

mmahadevan108, 2017-04-30:
I have noticed a USB issue when using ARM compiler. Its works with GCC, can you please try and confirm.

screamerbg, 2017-04-30:
@mmahadevan108 Most of the USB Device device users are using the mbed Online Compiler, which at its very heart is ARM Compiler. Could you please check with. ARMCC?

ashok-rao, 2017-05-10:
Hi @mmahadevan108 , I can confirm that issue is reproducible with both GCC and ARMCC. On Win10, 64-bit, Enterprise edition, I'm facing the following with K64F:

[image]

And device manager reports this:
[image]

ashok-rao, 2017-05-18:
Update on the above; Issue also observed with FRDM-K22F (USB device not recognized, same as for K64F) & FRDM-KL46Z (Board not even powering up from USB OTG port).

RdaZhongyao, 2017-05-19:
@screamerbg , I met the same problem while using `USBDevice/USBMSD` + `mbed-5.4.5 release`, with K64F board and ARMCC compiler.

After tacking the code, I believe that `malloc` is called in `usbisr` raises this bug.
See file : https://github.com/ARMmbed/mbed-os/blob/2e2fe377b3abc2620682dd7a0a98b9a32a5b2c87/features/unsupported/USBDevice/USBDevice/USBHAL_KL25Z.cpp
where we can find : `USBHAL::usbisr` -> `USBHAL::realiseEndpoint` -> `malloc`

Hope that helps.

mmahadevan108, 2017-05-19:
@RdaZhongyao Thanks.

maclobdell, 2017-07-24:
@screamerbg does the fix satisfy this issue?  Okay to close?

closed 2017-11-27 by maclobdell