ARMmbed/mbed-os #4257: UBLOX_EVK_ODIN_W2 - SD-card not working if WiFi is used

JanneKiiskila, 2017-05-02:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)
1.1.1

**meed-os sha:**
(`git log -n1 --oneline`)

42be5c0 (HEAD, tag: mbed_lib_rev140, tag: mbed-os-5.4.3) Merge pull request #413

**DAPLink version:**

**Expected behavior**

SD-card and WiFi should be able to co-exist.

**Actual behavior**

Once WiFi is activated, SD-card loses it's clocks.
Works with ethernet.

**Steps to reproduce**

Use mbed-cloud-client-example (or try to use WiFi and SD-card at the same, no suitable publicly available example available straight off the bat I'm afraid).

Ref: https://jira.arm.com/browse/IOTMORF-966

0xc0170, 2017-05-02:
cc @andreaslarssonublox

0xc0170, 2017-05-02:
cc @andreaspeterssonublox

JanneKiiskila, 2017-05-02:
I'm trying to write quickly a hack of the example1.cpp in sd-driver repo, which would reproduce this issue easily.

sg-, 2017-05-02:
There is a comment ```res``` https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_UBLOX_EVK_ODIN_W2/PinNames.h#L101 and also a definition for SCK https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_UBLOX_EVK_ODIN_W2/PinNames.h#L149

Does ```res``` mean reserved??

@andreaslarssonublox @screamerbg

0xc0170, 2017-05-03:
According to the w2 manual https://www.u-blox.com/sites/default/files/ODIN-W2_SIM_(UBX-14040040).pdf those pins are not used by the driver and should be usable: "SPI not used for ODIN connectivity software". However for odin board driver, there is a note in their docs that SDIO is used by wifi driver but not specified which one (is there just one?).

Not certain why it is as ``res`` ?

andreaslarssonublox, 2017-05-03:
Some of the pins SPI pins are re-initialized by the driver and it seems to work after fixing that. Need to do some more tests though. Will fix it in the next release of the driver.
Don't know why 'res' is used in PinNames.h since they should be possible to use.

0xc0170, 2017-05-03:
> Some of the pins SPI pins are re-initialized by the driver and it seems to work after fixing that. Need to do some more tests though. Will fix it in the next release of the driver.

Thanks for the support! Time frame for this fix? we got code-freeze for the next patch release in one day.

andreaslarssonublox, 2017-05-04:
We will not be able to finalize a release today so we aim for the next in two weeks instead.

0xc0170, 2017-05-04:
> We will not be able to finalize a release today so we aim for the next in two weeks instead. Would that be OK?

This should go to the known issues (cc @adbridge ). Please keep us updated

JanneKiiskila, 2017-05-15:
@andreaslarssonublox  - any chance of making the next mbed OS patch release? I'd really like to expand our board support for Cloud Client, so this fix would be essential.

adbridge, 2017-05-19:
@JanneKiiskila I assume this issue can now be closed then?

closed 2017-07-06 by sg-