ARMmbed/mbed-os #4261: STM32F4XX HAL: stopping ADC DMA can incorrectly return error

Pliny, 2017-05-02:
**Overview**
`HAL_ADC_Stop_DMA()` can incorrectly return `HAL_DMA_ERROR_NO_XFER` and may not properly stop the DMA engine.

**Target**
`stm32f4xx`. I found this on the [Nucleo_F446RE](https://developer.mbed.org/platforms/ST-Nucleo-F446RE/)

**other thoughts**
If there is an abort request while DMA is not busy, should it really be considered an ERROR? Maybe instead it should just return `HAL_OK`. Any thoughts?

Pliny, 2017-05-02:
Same behavior exists with the following
* stm32f7xx
* stm32l4xx
* stm32f2xx

The code in question is here: 
[stm32f4xx_hal_dma.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_dma.c#L523)

LMESTM, 2017-05-03:
@Pliny  When you mention that it may not properly stop the DMA, can you explicitly mention the DMA state and where this can happen ? Typically, what was the DMA HAL state when you've encountered the error ?

LMESTM, 2017-05-03:
I have also transferred your question to DMA team, I'll keep you posted

Pliny, 2017-05-03:
Hi @LMESTM , thanks for the quick response.

`__HAL_DMA_DISABLE(hdma)` will never get called unless there is an explicit `HAL_DMA_DeInit()` call or there's an error condition on interrupt. Currently, there's no clean way to disable the DMA without resetting the DMA engine.

Similarly, transfer related interrupts will not get disabled.

To me, it seems as if `HAL_DMA_Abort()` is treated as a _dma disable_ call in [stm32f4xx_hal_adc.c#L1154](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_adc.c#L1154) instead of a _conditional abort_ call.

I'm curious to know why this check exists [stm32f4xx_hal_dma.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_dma.c#L523).

If it needs to be there, maybe the conditional could be
```
if(hdma->State == HAL_DMA_STATE_READY)
```
This assumes `HAL_DMA_STATE_READY` states that the system is idle, yet enabled, and that interrupts are disabled.

Pliny, 2017-05-03:
Ah, I just realized I didn't really satisfy your request :)

you can set up a test with the following pseudocode
```
configure_clocks_and_pins()
configure_nonblocking_adc(&AdcHandle)
configure_dma(&DmaHandle)
__HAL_LINKDMA()
configure_nvic()
HAL_DMA_Init(&DmaHandle)
HAL_ADC_Init(&AdcHandle)

HAL_ADC_Start_DMA();
while(poll_for_mem0_ready()) {}
HAL_ADC_Stop_DMA();

HAL_ADC_Start_DMA(); /*< Here, you will get both a DMA and ADC error */
```

bcostm, 2017-05-18:
ST_INTERNAL_REF 33814

LMESTM, 2017-06-01:
@Pliny  Hi - our HAL team is asking if you could provide the ADC configuration that you are using ?

Pliny, 2017-06-01:
Hi, this issue went away with this checkin: https://github.com/ARMmbed/mbed-os/pull/4263/commits/42f66eec60abc2f750b7c7f4353d5b375d2f9aed

LMESTM, 2017-06-01:
@Pliny  ok thanks, then please close the issue

closed 2017-06-01 by Pliny