ARMmbed/mbed-os #4262: STM32F4XX: DMA HAL busy/ready conditionals are incorrect

Pliny, 2017-05-03:
The stm32f4xx has additional [states](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_dma.h#L121) as compared to other stm32 parts.

```
/** 
  * @brief  HAL DMA State structures definition
  */
typedef enum
{
  HAL_DMA_STATE_RESET             = 0x00U,  /*!< DMA not yet initialized or disabled */
  HAL_DMA_STATE_READY             = 0x01U,  /*!< DMA initialized and ready for use   */
  HAL_DMA_STATE_READY_MEM0        = 0x11U,  /*!< DMA Mem0 process success            */ // FIX
  HAL_DMA_STATE_READY_MEM1        = 0x21U,  /*!< DMA Mem1 process success            */ // FIX
  HAL_DMA_STATE_READY_HALF_MEM0   = 0x31U,  /*!< DMA Mem0 Half process success       */ // FIX
  HAL_DMA_STATE_READY_HALF_MEM1   = 0x41U,  /*!< DMA Mem1 Half process success       */ // FIX
  HAL_DMA_STATE_BUSY              = 0x02U,  /*!< DMA process is ongoing              */
  HAL_DMA_STATE_BUSY_MEM0         = 0x12U,  /*!< DMA Mem0 process is ongoing         */ // FIX
  HAL_DMA_STATE_BUSY_MEM1         = 0x22U,  /*!< DMA Mem1 process is ongoing         */ // FIX
  HAL_DMA_STATE_TIMEOUT           = 0x03U,  /*!< DMA timeout state                   */
  HAL_DMA_STATE_ERROR             = 0x04U,  /*!< DMA error state                     */
  HAL_DMA_STATE_ABORT             = 0x05U,  /*!< DMA Abort state                     */
}HAL_DMA_StateTypeDef;
```

These state enumeration values for `HAL_DMA_STATE_READY` and `HAL_DMA_STATE_BUSY` are set up as maskable flags.

However, [stm32f4xx_hal_dma.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_dma.c) assumes natural enumerations in conditional.

This introduces potential corner case issues relating to stopping, starting, init/deinit, and setting callbacks.

Pliny, 2017-05-03:
[stm32f1xx_hal_dma.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F1/device/stm32f1xx_hal_dma.h#L110)
 has the same issue for READY states

LMESTM, 2017-05-03:
@Pliny thanks for reporting the issue.
From what I've seen, your proposal makes sense.
I shared your report and proposed fix to our DMA team and will let you know when we get some feedback.

betzw, 2017-05-04:
Pls. refer to this [comment](https://github.com/ARMmbed/mbed-os/pull/4263#issuecomment-299153244) of mine!

LMESTM, 2017-06-05:
@Pliny @betzw  As the previous patch from @betzw has been reverted, shall we close this ticket as well ?

Pliny, 2017-06-05:
yes, should be closed

closed 2017-06-05 by Pliny