ARMmbed/mbed-os #4268: How to Monitor CPU Usage of Thread Like FreeRTOS

nightseas, 2017-05-03:
### Description
- Type: Question
- Related issue: N/A
- Priority: Minor

---------------------------------------------------------------

## Question

How to monitor Threads' CPU usage in mbed OS like FreeRTOS?
I know it's basically measuring the run time, But how to get the idle thread run time, and is there any other time should be counted to calculate the whole system run time?

Thanks.

For an OS it's quite an important feature to monitor the CPU usage/utilization.
Here's an example from FreeRTOS.
[image]

bulislaw, 2017-05-04:
I don't think there's funcionality like that in RTX4, for RTX5 that can be found in our feature branch https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5  and will be merged soon, you may be able to use events http://arm-software.github.io/CMSIS_5/RTOS2/html/group__rtx__evr.html in any case it's better to ask RTX guys here https://github.com/ARM-software/CMSIS_5/

SenRamakri, 2018-06-14:
@nightseas - Thanks for your question. As part of 5.9 release we have introduced Device Statistics APIs which may address some of the things you are looking for. Please look into below links for more details.

https://os.mbed.com/blog/entry/Mbed-OS-59-release/
https://os.mbed.com/docs/v5.9/reference/mbed-statistics.html

We will also take your suggestions and will try to analyze if any gaps are there with current implementation and may address the same in future. 
I'll also be closing this issue as there are no further actions required and hopefully we have answered your questions.

closed 2018-06-14 by SenRamakri

agausmann, 2024-03-12:
I'm confused why this was closed. A bunch of statistics were added (which are helpful in other scenarios), but as of 6.16 none of them cover the original topic of the issue, which is **per-thread CPU usage**

```c
typedef struct {
    us_timestamp_t uptime;            /**< Time since the system has started */
    us_timestamp_t idle_time;         /**< Time spent in the idle thread since the system has started */
    us_timestamp_t sleep_time;        /**< Time spent in sleep since the system has started */
    us_timestamp_t deep_sleep_time;   /**< Time spent in deep sleep since the system has started */
} mbed_stats_cpu_t;

typedef struct {
    uint32_t id;                /**< ID of the thread */
    uint32_t state;             /**< State of the thread */
    uint32_t priority;          /**< Priority of the thread (higher number indicates higher priority) */
    uint32_t stack_size;        /**< Current number of bytes reserved for the stack */
    uint32_t stack_space;       /**< Current number of free bytes remaining on the stack */
    const char   *name;         /**< Name of the thread */
} mbed_stats_thread_t;
```