ARMmbed/mbed-os #4269: Nucleo_blink_led fails to compile on SW4STM32

bsrirao, 2017-05-03:
### Description
- Type: Bug
- Priority: Blocker
   The Nucleo_blink_led exported project from mbed online fails to compile on SW4STM32 with the following error. I was reading in same issue reported for a different platform, but I am not sure how to set preprocess linker script to SW4STM32 tool chain.

`Description	Resource	Path	Location	Type
ignoring invalid character # in expression	Nucleo_blink_led line 13, external location: /Applications/Ac6/SystemWorkbench.app/Contents/Eclipse/plugins/fr.ac6.mcu.externaltools.arm-none.macos64_1.13.1.201703061524/tools/compiler/arm-none-eabi/bin/ld:/Users/test/Documents/workspace_stm32/Nucleo_blink_led/mbed/TARGET_NUCLEO_F429ZI/TOOLCHAIN_GCC_ARM/STM32F429xI.ld	C/C++ Problem
syntax error	Nucleo_blink_led		line 13, external location:/Applications/Ac6/SystemWorkbench.app/Contents/Eclipse/plugins/fr.ac6.mcu.externaltools.arm-none.macos64_1.13.1.201703061524/tools/compiler/arm-none-eabi/bin/ld:/Users/test/Documents/workspace_stm32/Nucleo_blink_led/mbed/TARGET_NUCLEO_F429ZI/TOOLCHAIN_GCC_ARM/STM32F429xI.ld`

---------------------------------------------------------------
## Bug

**Target**
STM32

**Toolchain:**
System Workbench for STM32 with GCC_ARM

**Toolchain version:**
1.14.0

**Expected behavior**
Project should compile successfully.

**Actual behavior**
Compilation fails with errors.

bsrirao, 2017-05-20:
Any idea when this will be getting fixed ?

theotherjimmy, 2017-05-22:
@bsrirao As a workaround, you could use the `gnuarmeclipse` or `eclipse_gcc_arm` exporters.

bsrirao, 2017-05-25:
Thanks for the answer @theotherjimmy. Are they work with STM32 System work bench IDE ?

theotherjimmy, 2017-05-25:
I think that question could be either:
1) is STM working on their export support? That's a question for the ST team, but I think the answer is yes.
2) Do the exporters that I listed as workarounds work with SW4STM32? I think so. `eclipse_gcc_arm` will work with all CDT-based eclipse IDEs and `gnuarmeclipse` will work with all GNU ARM Eclpise based IDEs. GNU ARM Eclipse depends on CDT, so it's a stronger requirement.

0xc0170, 2017-05-26:
cc @bcostm @adustm @LMESTM @jeromecoutant

JojoS62, 2017-06-26:
On my first trial to use the SW4STM32 exporter I ran also into a problem that maybe the same source. 
The setting for the symbol CMSIS_VECTAB_VIRTUAL_HEADER_FILE causes also a problem in the exporter. It adds the line
```
<listOptionValue builtIn="false" value="CMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h""/>
```
to the .cproject file and it is invalid. So the config is not read at all and eclipse complains about 'not a CDT project' in the settings dialog. Other symbol definitions work wll, but the special thing here that there is a string in definition. I don't now the correct syntax for this yet.

JojoS62, 2017-06-26:
this is what the exporter must generate for a string in an option:
<listOptionValue builtIn="false" value="CMSIS_VECTAB_VIRTUAL_HEADER_FILE=&quot;cmsis_nvic.h&quot;"/>
But I see that this is another issue as described by @bsrirao

LMESTM, 2017-07-19:
Hopefully ongoing fix here: #4779

ajaakko-arm, 2017-08-16:
Just pushed new changes to PR https://github.com/ARMmbed/mbed-os/pull/4779
That should fix linker script problem and 'ignoring invalid character #'-error.

bcostm, 2017-10-16:
@bsrirao Can you please check again and close this issue if there is no more problem. Thanks.

ST_TO_BE_CLOSED

closed 2018-01-24 by 0xc0170