ARMmbed/mbed-os #4283: Regarding SerialBase class

YarivCol, 2017-05-08:
### Description
- Type: Enhancement | Question

---------------------------------------------------------------
Hey!

Is there any reason why SerialBase class's methodes start_read, start_write, abort_read and abort_write arent virtual?

If they'll be it will be easier for me to imlplement my own types of serial comm...
Ill be also able to wrap serialBase functionality with my needs...

Thanks!

geky, 2017-05-12:
Not that I can tell, unless @sg- knows of a reason, feel free to put up a pr to change those functions to virtual declarations!

SenRamakri, 2018-06-19:
Hi @YarivCol - Thanks for your question and suggestions. Hopefully we responded with the best course of action on this. As @geky mentioned the best way to proceed on this is to raise a PR with those functions implemented as virtual so that you can get the changes reviewed formally.  For now, we will close this issue, but feel free to reopen if required.

closed 2018-06-19 by SenRamakri