ARMmbed/mbed-os #4285: Is there any plan to upgrade the ChaN FAT filesystem to R0.12 ?

RdaZhongyao, 2017-05-08:
### Description
- Type: Enhancement
- Related issue: `#FAT`
- Priority: Minor

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
- ChaN FAT filesystem supports `exFAT` since R0.12, the latest is R0.12c, mbed is using R0.11a
- `exFAT` supports storage device over 32GB

**Suggested enhancement**
- Upgrade to R0.12c

sg-, 2017-05-11:
We don't have a current plan to but would welcome the contribution.

infinnovation, 2017-11-15:
Note that exFAT is patented and its use [requires purchasing a licence from Microsoft](https://www.microsoft.com/en-us/legal/intellectualproperty/mtl/exfat-licensing.aspx)

0xc0170, 2018-06-15:
Resolved via 5490 (updated to the newer version)

closed 2018-06-15 by 0xc0170