ARMmbed/mbed-os #4287: Multithreading pyOCD debugging doesn't work on visual studio code

c1728p9, 2017-05-08:
Reporting at the request of @janjongboom 

When using a project exported to visual studio code with and pyOCD with RTOS integration (https://github.com/mbedmicro/pyOCD/pull/242) the debugging session is abruptly terminated due to thread ID 1 not existing. A log of this can be found here [visual_studio_log.txt](https://github.com/ARMmbed/mbed-os/files/984593/visual_studio_log.txt).

flit, 2017-05-13:
First, note that there are two types of thread IDs: the target thread ID, as reported by the gdb server, and gdb's own thread IDs. The latter start at 1 and increment for every new thread.

pyOCD uses a target thread ID of 1 as the default thread before RTOS integration is enabled. This will also end up with a gdb thread ID of 1 since it's the first thread reported by pyOCD.

I think this error is caused by thread ID 1 disappearing and being replaced with the RTOS threads. You can see this being reported here in the log:
```
1: (5528) <-1020-thread-info
1: (5593) ->=thread-exited,id="1",group-id="i1"
1: (5600) ->=thread-created,id="2",group-id="i1"
1: (5600) ->~"[New Thread -2146959359]\n"
1: (5600) ->=thread-created,id="3",group-id="i1"
1: (5600) ->~"[New Thread -2117580494]\n"
```

What I don't fully understand is who is reporting the error? Is it gdb, or Visual Studio Code?

There also seems to be some connection to having a breakpoint set on thread 1 based on the error message: "Failed to find thread 1 for break event".

Interestingly, on rare occasions I've seen Eclipse report an error about thread ID 1 not being found. But it hasn't been consistent enough that I could track it down. And it's not fatal to the debug session.

0xc0170, 2019-10-24:
@flit Shall this be moved to pyOCD? Is it still relevant?

0xc0170, 2019-10-25:
> You can only transfer issues between repositories owned by the same user or organization account.

I can't transfer this to pyOCD.  I'll close this and if you find this still relevant, please create it in pyOCD repository. Thanks!

closed 2019-10-25 by 0xc0170