ARMmbed/mbed-os #4288: ARCH_PRO - Flash - buffer alignment test is failing

studavekar, 2017-05-08:
ARCH_PRO - Flash - buffer alignment test is failing on master

- Type: Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
ARCH_PRO 

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-os sha:**
Merge: 3b11b23 7f12ad2
Author: Jimmy Brisson <theotherjimmy@gmail.com>
Date:   Mon May 8 11:16:31 2017 -0500

    Merge pull request #4149 from monkiineko/master

    STM32F3: Correct handling of USB ISTR and endpoint registers

**DAPLink version:**
0242

**Expected behavior**
Flash - buffer alignment test is failing tests should pass 

**Actual behavior**
Flash - buffer alignment test is failing tests should pass 

**Steps to reproduce**

- mbed  test --compile -m ARCH_PRO -t ARM

- mbedhtrun -m ARCH_PRO -p COM27:9600 -f BUILD/tests/ARCH_PRO/ARM/mbed-os/TESTS/mbed_hal/flash/flash.bin -d D:

studavekar, 2017-05-08:
@0xc0170 this issue was seen in CI, tried locally on different device still the tests failed.

0xc0170, 2017-05-09:
> @0xc0170 this issue was seen in CI, tried locally on different device still the tests failed.

Thanks for reporting

cc @0x6d61726b - can you have a look?

0xc0170, 2017-05-09:
I believe for now as ARCH PRO is failing, and affects all of us, I will disable flash algo support until it gets fixed

0xc0170, 2017-05-11:
I run local tests

```
+-------------+---------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target      | platform_name | test suite                   | test case                     | passed | failed | result | elapsed_time (sec) |
+-------------+---------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - buffer alignment test | 0      | 1      | FAIL   | 0.19
            |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - erase sector          | 1      | 0      | OK     | 0.14
            |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - init                  | 1      | 0      | OK     | 0.06
            |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - mapping alignment     | 1      | 0      | OK     | 0.06
            |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - program page          | 1      | 0      | OK     | 0.26
            |
+-------------+---------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
```

0xc0170, 2017-05-11:
Disabled now on master.

Reference the PR that introduced it https://github.com/ARMmbed/mbed-os/pull/4169

chrissnow, 2017-06-28:
@0xc0170 I'm looking into this for another bootloader project. it looks like the processor requires the buffer to be aligned in order for IAP to work. any suggestions other than having to memcpy to an aligned area if it isn't already aligned?, it would require another 1k of RAM which probably isn't an issue if ran inside a bootloader.

If the algo was alteredto use 256 byte pages this wouldn't be so bad.

0xc0170, 2017-06-28:
As I remember the algo were already modified, feel free to modify to make it work, if it does not work out of the box. Or implement flashiap by own driver , might be better..

chrissnow, 2017-06-28:
indeed they were, just looking through the diff that was posted.

I have already started implementing it myself as a flash api, is this the preferred method over the flash algo? certainly seems a bit more maintainable!

assuming that the buffer has to be aligned is memcpy the right thing to do? or should something else be considered? It seems like a bit of a bodge but if it's just an NXP thing I guess not so bad.

0xc0170, 2017-06-28:
> I have already started implementing it myself as a flash api, is this the preferred method over the flash algo? certainly seems a bit more maintainable!

Yes, it is ! 

If there is no other way and it is required to be aligned, then yes.

chrissnow, 2017-08-29:
work in progress here
https://github.com/chrissnow/mbed-os/tree/LPC1768-IAP-fix

```
+-------------+---------------+-------------------------------------+---------------------------+--------+--------+--------+--------------------+
| target      | platform_name | test suite                          | test case                 | passed | failed | result | elapsed_time (sec) |
+-------------+---------------+-------------------------------------+---------------------------+--------+--------+--------+--------------------+
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_drivers-flashiap | FlashIAP - init           | 1      | 0      | OK     | 0.04               |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_drivers-flashiap | FlashIAP - program        | 1      | 0      | OK     | 0.29               |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_drivers-flashiap | FlashIAP - program errors | 1      | 0      | OK     | 0.06               |
+-------------+---------------+-------------------------------------+---------------------------+--------+--------+--------+--------------------+
```
```
+-------------+---------------+------------------------------+---------------------------+--------+--------+--------+--------------------+
| target      | platform_name | test suite                   | test case                 | passed | failed | result | elapsed_time (sec) |
+-------------+---------------+------------------------------+---------------------------+--------+--------+--------+--------------------+
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - init              | 1      | 0      | OK     | 0.07               |
| LPC1768-ARM | LPC1768       | mbed-os-tests-mbed_hal-flash | Flash - mapping alignment | 0      | 1      | FAIL   | 0.1                |
+-------------+---------------+------------------------------+---------------------------+--------+--------+--------+--------------------+
```

chrissnow, 2017-08-30:
PR #4993 Should resolve this.

closed 2017-09-25 by 0xc0170