ARMmbed/mbed-os #4297: The online IDE is stucked at mbed-os 5.4.3

vargham, 2017-05-10:
### Description
- Type: Bug
- Priority: Major
The online IDE is stucked at mbed-os 5.4.3, could not use any other version.
I can change it in the revision screen, but it has no effect.
[image]

---------------------------------------------------------------
## Bug

**Target**
All

**Steps to reproduce**
Import mbed OS Blinky https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-blinky/rev/bb184e64b703 Update all libraries.
Add:
`printf("MBED version: %d.%d.%d (lib %d)\r\n", MBED_MAJOR_VERSION, MBED_MINOR_VERSION, MBED_PATCH_VERSION, MBED_LIBRARY_VERSION);`
Itt will always print:
`MBED version: 5.4.3 (lib 140)`

theotherjimmy, 2017-05-10:
have you tried rebuilding from scratch? you have to click on the down arrow next to compile, and then click "compile all"

vargham, 2017-05-10:
Why compile all? It is a freshly imported, official mbed-os blinky. Anyway I tried compile all. The result is: `MBED version: 5.4.4 (lib 141)` Strange.

vargham, 2017-05-10:
I also exported the project to offline (ARM gcc makefile).
At the issue creation time it exported 5.4.3. Now I tried again and it exported 5.4.4
The Revision History still shows 5.4.5

theotherjimmy, 2017-05-10:
@vargham Your right, that's really weird. You just imported mbed-os-example-blinky? I'll see if I can reproduce this.

theotherjimmy, 2017-05-10:
Thanks for the details. I'll give this a go.

theotherjimmy, 2017-05-10:
I just did a fresh import, copied your single printf into main, and got: `MBED version: 5.4.5 (lib 142)`

theotherjimmy, 2017-05-10:
I think that's correct

vargham, 2017-05-10:
I tried again. (Did not touch anything.) The build and the offline export are both OK now.
I compared the previous two bad exports' files to the github repository and the exported projects really contained the 4.5.3 and 4.5.4 version.

theotherjimmy, 2017-05-10:
Probably a typo for 5.4.3 and 5.4.5 right? otherwise that's very strange O_o

vargham, 2017-05-10:
Typo? Whose?
1. I used the online IDE with freshly imported mbed-os blinky.
2. I checked the revision history, the screenshot is here, it showed 5.4.5.
3. Built online. The printed version was 5.4.3
4. Exported to offline. The mbed-os files were version 5.4.3
5. Waited...
6. Built online. The printed version was 5.4.4
7. Exported to offline. The mbed-os files were version 5.4.4
8. Waited...
9. Built online. The printed version is 5.4.5 now
10. Exported to offline. The mbed-os files are version 5.4.5 now

theotherjimmy, 2017-05-10:
I was referring to this: 
> the 4.5.3 and 4.5.4 version.

Not anything else.

closed 2017-05-11 by vargham